TheThingsNetwork/lorawan-stack #2326: Inconsistent translations for stack component names in the Console

bafonins, 2020-04-09:
#### Summary

We should decide whether we want to translate stack component names in the Console.

#### Why do we need this?

Consistency

#### What is already there? What do you see now?

Currently, we do translate stack component names in the console overview:
https://github.com/TheThingsNetwork/lorawan-stack/blob/master/pkg/webui/console/views/overview/index.js#L65-L71

But in the device import page we dont: 
https://github.com/TheThingsNetwork/lorawan-stack/blob/master/pkg/webui/constants/components.js#L15-L23

#### What is missing? What do you want to see?

Consistent definition of stack component names in the web ui

#### Environment

v3.7.0

#### How do you propose to implement this?

According to https://github.com/TheThingsNetwork/lorawan-stack/pull/2253#issuecomment-605402188 we dont want to translate those, so we should remove translations from the overview page.

@johanstokking Can you explain why is it so? Also, if we dont internationalize, for example, `Network Server` how the translation for `Network Server Address` would look?

Additionally, I can see that many sales pages/articles those are adapted to other languages than english

#### Can you do this yourself and submit a Pull Request?

no

johanstokking, 2020-04-09:
I think the component names would be fixed; those are part of the product branding.

It's like translating Microsoft Internet Explorer to Microsoft интернет-исследователь. So "Application Server" is part of our product branding, it's not a description of what it is.

So it becomes something like "Network Server адрес"

Or would you say that we need to make this phonetic? Like Нетворк Сервер ?

bafonins, 2020-04-09:
Here are some examples from google image search when querying for lorawan architecture/lorawan network server in russian (there are much more):

[image]

A similar situation is with the Application Server.

Im fine with any decision though, just want the usage of stack component names to be consistent in the app.

rvolosatovs, 2020-04-09:
I think it makes the most sense to translate the component names to respective languages, our product is called "The Things Stack", that's something that stays and should not be translated, however having terms like "Network Server", "Identity Server", "Packet Broker" printed in English in the console just does not make sense in my opinion, here's why:
1. first and foremost, Network Server, Application Server, Join Server are all LoRaWAN spec terms, not names of our product offerings, if they were, we would write them as e.g. "The Things Stack Join Server address", in which case I would agree for keeping the English name in, however in case of, for example, "Join Server address" - Join Server refers to the actual LoRaWAN spec-defined Join Server, which may or may not, in fact, be the Join Server we offer ("The Things Stack Join Server") or any other LoRaWAN-compliant Join Server.
2. There's no way of mixing "Network Server" and "адрес" without it looking as we just used Google Translate to translate it. Note, again, that there's nothing wrong with `адрес "The Things Stack Network Server"`.

I think it all depends on the context and IMO if we're specifically talking about our implementation of the component defined by the spec, we should use full English name("The Things Stack <component>"), if we're talking about just a generic LoRaWAN component, we should translate.
Note, that in result we may have both translated and non-translated terms in the text, e.g. `Сетевой сервер "The Things Stack Network Server"`.

kschiffer, 2020-04-10:
Differentiating between the LoRaWAN spec terms and our brand names is a fair point and we should do that I guess. But even there, if I think about, say German translation, I really wouldn't want `Application Server` to be called `Anwendungsserver` and I think a lot of people would feel that way. Then again, in e.g. French, the sentiments might be different.
In the end though, I think we can go ahead and localize LoRaWAN spec terms and let the translators decide the specifics.

As @rvolosatovs already mentioned we should then also think about how to communicate the difference between our brand offerings and the LoRaWAN spec terms. `"The Things Stack Network Server"` is I guess our only bet here.

johanstokking, 2020-04-10:
> say German translation, I really wouldn't want `Application Server` to be called `Anwendungsserver` and I think a lot of people would feel that way.

Yeah so this touches really the core of the issue. Do we want this, or not.

So is Network Server a generic term, or is it part of our branding. I think this even depends on the context; are we talking about our Network Server product, or are people configuring the address of a Network Server in the Join Server (which could also be another NS).

I think that we should make this translatable, and let it depend on the locale if we translate the terms or not. In Dutch, we wouldn't translate it I think, German neither, but in French and Russian it might be.

PR #2746 (2020-06-18): Stack component translations

#### Summary

Closes #2326 

#### Changes

- Add translations for stack components

#### Testing

Trivial

##### Regressions

None

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API, storage, configuration and CLI, according to the compatibility commitments in `README.md`. The target branch is set to `master` if the changes are fully compatible with existing API, database, configuration and CLI.
- [ ] Documentation: Relevant documentation is ad [... cut]
closed 2020-06-18 by pgalic96