TheThingsNetwork/lorawan-stack #2327: Sort events by `time` in console

rvolosatovs, 2020-04-09:
#### Summary

Console should sort events by `time` field

#### Why do we need this?

To give sensible log data to the users.

#### What is already there? What do you see now?

Seems that events are printed in the same order they arrive, which is not necessarily correct and confuses the user.

[image]

#### What is missing? What do you want to see?

Events sorted by `time` in the console for clarity.

#### How do you propose to implement this?

Let `A` be the list of events presented in the console, let `e` be the event received.
If `tail(A).time <= e.time`, `A = A ++ e`, repeat with next `e`.
Otherwise, use binary search to find the index at which to insert `e` and do so. Note, it may be beneficial to only search over `n` last elements, where `n<len(A)`, since in most cases, the insertion index `i` will be close to the tail.
Note(2), just looping in reverse order over `A` in order to find `i` is most probably also fine, however, once we start batching events and introduce more and more events, this may become problematic and using `log(n)` search algorithm may prove to me beneficial, it's also very straight-forward to implement. We could start do binary search over last 32 events perhaps and if not found, go to next "chunk" of events.
We should probably consider limiting the amount of events presented in the console as well, which can influence the choice of algorithm and it's parameters.

#### Can you do this yourself and submit a Pull Request?

I could, but I guess @bafonins is better suited for the job.

rvolosatovs, 2020-04-09:
cc @johanstokking @htdvisser for discussion on any other approaches we could take here (e.g. to make this server-side etc.), IMO it makes sense that clients have to sort themselves, Event Server will not(and should not) ensure this sorting.

htdvisser, 2020-04-09:
We're in a distributed environment, IS/GS/NS/AS/JS instances may be deployed on separate hosts, with different latencies to the events pubsub. This means that events are never guaranteed to arrive in order.

This is even more clear when the IS is in a different region, and the Console needs to open multiple event streams. No way that you're going to receive events in order then

KrishnaIyer, 2020-04-09:
We could attach a unix timestamp to the events when they occur and sort using that?

rvolosatovs, 2020-04-09:
> We could attach a unix timestamp to the events when they occur and sort using that? This would mean that events need to be re-sorted when there are new events.

That's what we already do and the approach is what I suggest in the OP

johanstokking, 2020-04-10:
I think sorting by time in the Console is not a bad idea. We won't be able to receive events in order, but we can try to present them in order. Of course the system clocks of the servers wouldn't be perfectly in sync, but we it would provide for a best effort sorting mechanism that works pretty well.

closed 2020-04-21 by bafonins