TheThingsNetwork/lorawan-stack #2351: Snap ttn-lw-stack executable runs the CLI instead

neoaggelos, 2020-04-11:
#### Summary

In a clean snap installation, `ttn-lw-stack` runs the CLI instead.

#### Steps to Reproduce

1. `sudo snap install ttn-lw-stack`
2. `ttn-lw-stack`

#### What do you see now?

`ttn-lw-stack` runs the CLI instead:

```
$ ttn-lw-stack version
The Things Network Command-line Interface: ttn-lw-cli
Version:             3.7.0
Build date:          2020-04-02T11:40:54Z
Git commit:          3c5048bc9
Go version:          go1.14.1
OS/Arch:             linux/amd64
```

Interestingly, `/snap/bin/ttn-lw-stack` is the cli, but `/snap/ttn-lw-stack/181/ttn-lw-stack` is the stack.

#### What do you want to see instead?

`ttn-lw-stack` should execute the stack, not the CLI.

#### Environment

3.7.0, snap revision 181. Cannot check with earlier revisions to see when this started.

#### How do you propose to implement this?

Fix our goreleaser or snapcraft configuration.

#### Can you do this yourself and submit a Pull Request?

Assigning this on @rvolosatovs. Re-assign if needed.

rvolosatovs, 2020-04-14:
@neoaggelos assigning this to you, please let me know if you need assistance with `goreleaser` - especially if you run into any weird bugs (I have on multiple occasions...).

neoaggelos, 2020-04-16:
It appears that this is actually a bug in `goreleaser`, which was introduced by this commit: https://github.com/goreleaser/goreleaser/commit/eb528971b9da279084886d2dba5f47bbd6df8fdd

The offending line is this one: https://github.com/goreleaser/goreleaser/blob/81417ea989d5a7635d7c3f9177b4ddf5b259f041/internal/pipe/snapcraft/snapcraft.go#L236

which ignores the application name. In our case, this overwrites `ttn-lw-stack` config `ttn-lw-cli` executable files. If we switch the order of the `build` list, then we get both `ttn-lw-cli` and `ttn-lw-stack` running the Stack instead.

I verified that this is an issue with goreleaser by building the snap with a version older than the commit: https://github.com/goreleaser/goreleaser/tree/v0.124.1 and the snap package builds OK.

This needs to be fixed in `goreleaser`.

neoaggelos, 2020-04-17:
Relevant PR in goreleaser has been merged upstream. https://github.com/goreleaser/goreleaser/pull/1449

Once new release is out, we will need to update the version we use in `go.mod`.

neoaggelos, 2020-04-20:
Goreleaser v0.132.0 is out, containing our fix.

PR #2376 (2020-04-21): all: Update goreleaser to v0.132.0

#### Summary

Closes #2351

#### Changes

- Update goreleaser version to v0.132.0, containing fix for snap packages

#### Notes for Reviewers

Not sure if this belongs in the changelog.

Tested that:
- snap packages are now OK
- `go run github.com/goreleaser/goreleaser --snapshot` works without errors.

#### Checklist

- [X] Scope: The referenced issue is addressed, there are no unrelated changes.
- [X] Compatibility: The changes are backwards compatible with existing API, database and configuration, according to the stability commitments in `README.md`.
- [ [... cut]
closed 2020-04-21 by KrishnaIyer