TheThingsNetwork/lorawan-stack #2355: Snap generating config file fails because of missing directory.

neoaggelos, 2020-04-12:
#### Summary

`ttn-lw-cli use <host> --user`  fails when CLI is installed using snap.

#### Steps to Reproduce

1. `sudo snap install ttn-lw-stack`
2. `ttn-lw-stack.ttn-lw-cli use localhost --used`

#### What do you see now?

```
 $ ttn-lw-stack.ttn-lw-cli use localhost --user
error:cmd/ttn-lw-cli/commands:fail_write (failed to write `HOME/snap/ttn-lw-stack/181/.config/.ttn-lw-cli.yml`)
    file=HOME/snap/ttn-lw-stack/181/.config/.ttn-lw-cli.yml
--- open HOME/snap/ttn-lw-stack/181/.config/.ttn-lw-cli.yml: no such file or directory
```

This happens because there is no directory `.config` directory inside the snap folder. If you create the directory by hand then it works.

#### What do you want to see instead?

The command to succeed (somehow)

#### Environment

3.7.0

#### How do you propose to implement this?

Two ideas:

- Create the destination directory if it does not exist already (not in favor of this one)
- When running in snap, add an extra default config file path `$SNAP_USER_COMMON` (which would be `HOME/snap/ttn-lw-stack/common`). Also, write configuration file there instead of the user config dir, which changes in every revision.

#### Can you do this yourself and submit a Pull Request?

yes, but let's discuss what's best first @htdvisser, @rvolosatovs

rvolosatovs, 2020-04-14:
I think there are 2 different issues here:
1. Creating the destination path to the config if it does not exist. I'm largely in favor of this. In fact it's extremely annoying to use CLI programs, which do not do this.
2. We should indeed write the config to a stable location, which would persist across versions - that's a higher priority than (1.), because if this does not happen, then the snap package is quite unusable as it is now IMO.

neoaggelos, 2020-06-20:
Closed with #2361

PR #2458 (2020-05-03): CLI create user config dir if missing

#### Summary

Closes #2355 

#### Changes

- Create user config directory if it does not exist, so that the config file generation does not fail.

#### Notes for Reviewers

Do we need a flag for disable this? e.g. `--no-make-dirs`

#### Checklist

- [X] Scope: The referenced issue is addressed, there are no unrelated changes.
- [X] Compatibility: The changes are backwards compatible with existing API, storage, configuration and CLI, according to the compatibility commitments in `README.md`. The target branch is set to `master` if the changes are fully com [... cut]
PR #2361 (2020-06-18): cli: Use SNAP_USER_COMMON for config files if available

#### Summary

Closes #2355

#### Changes

- Write configuration files in `SNAP_USER_COMMON` directory. This is so that config is using the common snap directory instead of the revision specific one.
- Read config from `SNAP_USER_COMMON` as well, if that is available.  

#### Notes for Reviewers

We should probably be doing something similar with the cache as well.

#### Checklist

- [X] Scope: The referenced issue is addressed, there are no unrelated changes.
- [X] Compatibility: The changes are backwards compatible with existing API, da [... cut]
closed 2020-06-20 by neoaggelos