TheThingsNetwork/lorawan-stack #2358: Restructure docs

benolayinka, 2020-04-14:
#### Summary

Change doc structure from
- Guides
- Concepts
- Components
- Reference

to
- Getting started
- Devices
- Gateways
- The Things Stack
- Integrations
- Reference

#### Why do we need this ?

Restructure docs to
- Reflect how users search for help
- Flatten hierarchy
- Match conceptual flow of data
- Add some useful sections

#### What is already there? What do you see now?

#### Current structure

- Guides
	- getting started
		- installation
		- certs
		- configuration
		- running the tts stack
		- console
			- create gateway
			- create app
			- create end device
		- command-line interface
		- working with data
	- connecting gateways
	- basic user management
	- working with events
	- downlink queue operations
	- class c and multicast
	- generate qr code
	- web ui branding
- Concepts
	- end device templates
	- email templates
	- webhook templates
- Components
	- identity server
	- gateway server
	- network server
	- application server
	- join server
	- console
	- qr code generator
	- command-line interface
	- device template converter
	- gateway configuration server
- Reference
	- api
	- configuration
	- application server data
	- application packages
	- email templates
	- end device templates
	- webhook templates
	- interoperability repo
	- networking

#### What is missing? What do you want to see?

#### Proposed structure

- Getting started
	- installation
		- certs
		- configuration
		- running the tts stack
	- console
		- create gateway
		- create app
		- create end device
	- command-line interface
	- working with data
	- connecting gateways
	- basic user management
	- working with events
	- downlink queue operations
	- class c and multicast
	- generate qr code
	- web ui branding
- Devices
	- device registration
	- working with data
	- device templates
- Gateways
	- connect a gateway
	- common gateways
- The Things Stack
	- components
	- cli
	- console
	- ...
- Integrations
	- webhook templates
	- email templates
	- mqtt
	- data formats
	- application packages
- Reference
	- api
	- configuration
	- interoperability
	- glossary
	- port allocations

#### How do you propose to document this?

This is a change to documentation.

#### Can you do this yourself and submit a Pull Request?

Yes. Feedback appreciated, especially @htdvisser @johanstokking @wienke @laurensslats

johanstokking, 2020-04-14:
I think this is a good suggestion. We can still do this since we haven't been publishing the docs to stable URLs yet.

benolayinka, 2020-04-16:
> I think this is a good suggestion. We can still do this since we haven't been publishing the docs to stable URLs yet.

i'll create a branch with the changes then. besides fixing internal links, is there anything i should look out for?

@kschiffer @pgalic96 is this going to affect anything on your end?

benolayinka, 2020-04-16:
Proposition to drop (or rename) 'The Things Stack,' nav menu
- Basic usage is all covered in 'Getting started'
- It's confusing as a nav option PLUS the title of the page (clicking the logo takes you home, clicking the nav item takes you to a sub menu)
- Components subsection can move to reference
- Gives visual space between Reference and Forum nav items

With:
[image]

Without:
[image]

@kschiffer @johanstokking

johanstokking, 2020-04-16:
I agree

kschiffer, 2020-04-17:
> @kschiffer @pgalic96 is this going to affect anything on your end?

No. We didn't use any links yet, so that's fine.

PR #2368 (2020-04-24): doc: Restructure

#### Summary

Closes #2358 

#### Changes

Change doc structure from
- Guides
- Concepts
- Components
- Reference

to
- Getting started
- Devices
- Gateways
- Integrations
- Reference

#### Notes for Reviewers

Restructure docs to
- Reflect how users search for help
- Flatten hierarchy
- Match conceptual flow of data
- Add some useful sections

Removed The Things Stack section referenced in #2358 because it made navigation confusing and was unnecessary. 

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [ ] Compatibility: The c [... cut]
closed 2020-04-28 by benolayinka