TheThingsNetwork/lorawan-stack #2359: Add glossary

benolayinka, 2020-04-14:
#### Summary

Add glossary to docs with terms/abbreviations and definitions.

#### Why do we need this ?

We can make these terms available as context tooltips in the console, so users can see the definitions of fields when they are adding devices, gateways, applications, etc.

#### What is already there? What do you see now?

No glossary.

#### What is missing? What do you want to see?

A glossary section under the reference section of docs. Hover-over tooltips in the console providing definitions for terms.

[image]

[image]

[image]

#### How do you propose to document this?

This is an addition to the documentation.

#### Can you do this yourself and submit a Pull Request?

Yes, need help from @kschiffer @pgalic96 for markdown templates and link to docs from console.

pgalic96, 2020-04-14:
Linking docs from console is already an issue (see #2254 )

kschiffer, 2020-04-15:
> Linking docs from console is already an issue (see #2254 )

Yeah, this is a separate issue, where we want to (maybe automatically) connect text in the console with entries from the glossary of the docs.

benolayinka, 2020-04-24:
pushed branch `issue/#2359-add-glossary` with an unstyled `/reference/glossary`

data in `doc/data/glossary/glossary.yml`

added `term`, `term-alt`, and `term-abbr` classes to the partial at `doc/themes/tts/layouts/partials/proto/glossary-term.html` so we can style them.. anyone with an opinion about how to css those up please chime in @htdvisser @kschiffer  @pgalic96 @laurensslats @pierrephz

PR #2425 (2020-05-04): doc: Add glossary

#### Summary

Closes #2359 

#### Changes

- Add markdown doc `doc/content/reference/glossary/_index.html`
- Add yaml data for glossary `doc/data/glossary/glossary.yaml`
- Add css for glossary dl, dt, and dd
- Add shortcode and partial to display glossary

#### Notes for Reviewers

The shortcode, partial, and css can definitely be improved. This is a minimal working push to get the glossary live.

The  comment doesn't hide the terms in the glossary display

[image]

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatib [... cut]
closed 2020-05-05 by benolayinka