TheThingsNetwork/lorawan-stack #2371: Invalid band channel data rate limits

rvolosatovs, 2020-04-17:
#### Summary

Some of band channel data rate limits are invalid

#### Steps to Reproduce

1. Look at https://github.com/TheThingsNetwork/lorawan-stack/blob/master/pkg/band/au_915_928.go#L42-L43

#### What do you see now?

Invalid DR limits in AU915

#### What do you want to see instead?

Valid limits in all bands

#### How do you propose to implement this?

Validate the limits and fix inconsistencies across all specs

#### Can you do this yourself and submit a Pull Request?

yes

rvolosatovs, 2020-04-17:
Setting `prio/medium`, because without this in some bands NS does not accept uplinks

htdvisser, 2020-04-17:
This needs to be fixed in https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.7.0/pkg/band/downgrades.go#L54-L61 as the current DRs are valid for PHY 1.0.2rA, but not for 1.0.2rB.

PR #2380 (2020-04-21): Fix various band inconsistencies

#### Summary

Closes #2371 

#### Changes

- Fix various inconsistencies in various bands, namely:
	- Incorrect uplink channel frequencies
	- Incorrect channel data rate limits
	- Incorrect TxPower offsets
	- Incorrect MaxEIRP values
	- Incorrect beacon data rates
- Use typed data rate enums

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API, database and configuration, according to the stability commitments in `README.md`.
- [x] Testing: T [... cut]
closed 2020-04-21 by rvolosatovs