TheThingsNetwork/lorawan-stack #2373: Document update procedure (Docker Compose)

jpmeijers, 2020-04-19:
#### Summary

I'm running The Things Stack V3.6.0 and want to update to V3.7.0. It's running using the provided docker-compose file with customisation as documented in the installation guide (https://thethingsstack.io/v3.7.0/guides/getting-started/installation/).

#### Why do we need this ?

It's unclear what procedure should be followed to to an update, as rumour has it that a database migrations is necessary. Therefore it is not a simple change of the docker image version in the compose file.

See https://github.com/TheThingsNetwork/lorawan-stack/issues/2284

#### What is already there? What do you see now?

The documentation explains very well how to do a new clean installation of the stack.

#### What is missing? What do you want to see?

I would like to see a step by step guide how to perform a version upgrade.

#### How do you propose to document this?

As another guide on https://thethingsstack.io/v3.7.0/guides/getting-started/installation/

#### Can you do this yourself and submit a Pull Request?

I'm not sure how to do the upgrade yet. And perhaps there are a couple of steps to follow to prevent data loss or corruption.

jpmeijers, 2020-04-19:
The steps I followed are as follows:

Change docker image version numbers in docker-compose file
```
nano docker-compose.yml 
```
Pull new images:
```
root@ttnmapper-thingsstack3:~# docker-compose pull
Pulling cockroach ... done
Pulling redis     ... done
Pulling stack     ... done
```
Shut down the stack
```
root@ttnmapper-thingsstack3:~# docker-compose down
Stopping root_stack_1     ... done
Stopping root_cockroach_1 ... done
Stopping root_redis_1     ... done
Removing root_stack_1     ... done
Removing root_cockroach_1 ... done
Removing root_redis_1     ... done
Removing network root_default
```
Run migrations
```
root@ttnmapper-thingsstack3:~# docker-compose run --rm stack is-db migrate
Creating network "root_default" with the default driver
Creating root_cockroach_1 ... done
Creating root_redis_1     ... done
  INFO Connecting to Identity Server database...
  INFO Detected database CockroachDB CCL v19.2.6 (x86_64-unknown-linux-gnu, built 2020/04/06 18:05:31, go1.12.12)
  INFO Migrating tables...                     
  INFO Successfully migrated                   
```
Start stack
```
root@ttnmapper-thingsstack3:~# docker-compose up -d
root_cockroach_1 is up-to-date
root_redis_1 is up-to-date
Creating root_stack_1 ... done
root@ttnmapper-thingsstack3:~# 
```

Flush UDP forwarding table
```
conntrack -D -p udp
```

johanstokking, 2020-04-20:
Thanks, also for the suggestions!

You are right, we need a page for updating TTS that has been installed via the Getting Started.

PR #2385 (2020-04-29): doc: Add upgrade instructions

#### Summary

Fixes #2373 

#### Changes

- Add `Upgrading` section to docs

#### Notes for Reviewers

Is it okay to include output of terminal in bash instructions? It can be helpful but we don't typically do it.

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API, database and configuration, according to the stability commitments in `README.md`.
- [ ] Testing: The changes are covered with unit tests. The changes are tested manually as well.
- [... cut]
closed 2020-04-30 by benolayinka