TheThingsNetwork/lorawan-stack #2379: Data/Event Stream not working

wasn-eu, 2020-04-20:
#### Summary
The Data and Events boxes are not working.
There is only an error message:

Cannot show events
An unknown error occurred. Please try again later.
refreshRestart Stream

#### Steps to Reproduce
v3.7.0 clean install with docker-compose on debian 10.3

#### What do you see now?
Cannot show events
An unknown error occurred. Please try again later.
refreshRestart Stream

#### What do you want to see instead?
 the latest events and data from the device, for the application or from the gateway

#### Environment
docker-compose with v3.7.0
debian 10.3

bafonins, 2020-04-26:
Can you try restarting the events stream? You can do this by the `Restart stream` title on the error notification:
[image]

wasn-eu, 2020-04-27:
if i click on Restart stream nothing happens.

wasn-eu, 2020-05-05:
how can i help to fix this problem?
can i look at any specific log file or something else?

wasn-eu, 2020-05-05:
when i open the data tab for a node i can see this in docker-compose logs:

```
stack_1      |   INFO Request handled                          duration=631.934µs method=GET namespace=web remote_addr=192.168.88.247:50738 request_id=01E7JDDB7S0AREFVVTFP7WCFMP response_size=2167 status=200 url=/console/applications/modularnode/data
stack_1      |   INFO Request handled                          duration=206.993815ms method=POST namespace=web remote_addr=192.168.88.247:50744 request_id=01E7JDDBSEM7T32142GKB7DWN7 response_size=275 status=200 url=/oauth/token
stack_1      |   INFO Request handled                          duration=315.321353ms method=GET namespace=web remote_addr=192.168.88.247:50742 request_id=01E7JDDBP4BJ9DZGNVSV7NDH8C response_size=160 status=200 url=/console/api/auth/token
stack_1      |   INFO Finished unary call                      duration=55.79388ms grpc_method=AuthInfo grpc_service=ttn.lorawan.v3.EntityAccess namespace=grpc request_id=01E7JDDC0V699MQ9QV2RRE0GWD
stack_1      |   INFO Finished unary call                      duration=60.001764ms grpc_method=Get grpc_service=ttn.lorawan.v3.UserRegistry namespace=grpc request_id=01E7JDDC356GN209XE7E5PRVD3
stack_1      |   INFO Finished unary call                      duration=99.865752ms grpc_method=ListRights grpc_service=ttn.lorawan.v3.ApplicationAccess namespace=grpc request_id=01E7JDDC6198EBS3R53DD2V0W6
stack_1      |   INFO Finished unary call                      duration=101.440169ms grpc_method=Get grpc_service=ttn.lorawan.v3.ApplicationRegistry namespace=grpc request_id=01E7JDDC6095N1ZD9PHHVDNWZ2
stack_1      |   INFO Finished streaming call                  duration=1m0.059025624s error=context canceled grpc_method=Stream grpc_service=ttn.lorawan.v3.Events namespace=grpc request_id=01E7JDDC9JV3RYRCEZEKVR2G1C
```

wasn-eu, 2020-05-05:
after the "INFO Finished streaming call" i get this message in the browser:

```
Cannot show events
An unknown error occurred. Please try again later.
refresh Restart Stream 
```

pressing Restart Stream does nothing

bafonins, 2020-05-05:
> if i click on Restart stream nothing happens.

Hm, then most probably something is wrong with the configuration of the stack. Could you check the console in developer tools of your browser for any errors?

wasn-eu, 2020-05-05:
[image]
i get this.
trying to open https://ttn.wasn.eu/api/v3/events
i get:

`{"code":12,"message":"Not Implemented"}`

wasn-eu, 2020-05-05:
click on restart stream just starts a new fetch of events but with the same timeout

bafonins, 2020-05-05:
I see, could you share your configuration for `lorawan-stack` ?

Also, do you get the same response when using the cli? you can find how to set it up [here](https://thethingsstack.io/v3.7.0/guides/getting-started/cli/). Running
```
ttn-lw-cli events --application-id <your-app-id>
```
should open the events stream.

wasn-eu, 2020-05-05:
when i try the login i open the url in an browser and past the code in the terminal i get this error:

```
ERROR Could not exchange OAuth access token    error=Post "https://ttn.wasn.eu/oauth/token": remote error: tls: internal error
Post "https://ttn.wasn.eu/oauth/token": remote error: tls: internal error
```

wasn-eu, 2020-05-05:
tried it again from a linux machine:

```
INFO Got OAuth access token                  
root@dev:~# /snap/bin/ttn-lw-stack.ttn-lw-cli events --application-id modularnode
  WARN grpc: addrConn.createTransport failed to connect to {ttn.wasn.eu:8884  <nil> 0 <nil>}. Err: connection error: desc = "transport: authentication handshake failed: context deadline exceeded". Reconnecting...
```

wasn-eu, 2020-05-05:
got the cli to work now.
and i can see the data and events.
but still not in the webui

wasn-eu, 2020-05-05:
circled the problem.

i think that it is regarding tls.
because the ttn server is behind an nginx proxy on another server the lets encrypt certificate could not be generated.

but i am totaly stuck at using my existing certificate.
i uncommented all in yml files for signed certificates and commented out the tls part for lets encrypt. but now i get an error that the pem files are not found

neoaggelos, 2020-05-05:
> circled the problem.
>
> i think that it is regarding tls.
> because the ttn server is behind an nginx proxy on another server the lets encrypt certificate could not be generated.
>
> but i am totaly stuck [...]

Hi,

If you are using nginx as a reverse proxy in front of the stack, then nginx should be handling the TLS termination, and in turn use the HTTP port of the stack (1885).

wasn-eu, 2020-05-05:
nginx is using the 1885 port.
if i open the url for the events directly i get this xml:

```

error |  
-- | --
grpc_code | 7
http_code | 403
message | "error:pkg/auth/rights:no_application_rights (no rights for application `modularnode`)"
http_status | "Forbidden"
details |  
0 |  
@type | "type.googleapis.com/ttn.lorawan.v3.ErrorDetails"
namespace | "pkg/auth/rights"
name | "no_application_rights"
message_format | "no rights for application `{uid}`"
attributes |  
uid | "modularnode"
correlation_id | "838f26a01b024fbdb01f6ae83f517964"
code | 7

```

wasn-eu, 2020-05-06:
got it to work.
was a little work with nginx now handling all ssl for all ports.

but it still break very often

bafonins, 2020-05-06:
> got it to work.

Nice to hear!

> was a little work with nginx now handling all ssl for all ports.

It would be great if you can summarize what was wrong in your setup such that other users that experience the same issue can fix it. We might even include it in the docs (https://github.com/TheThingsNetwork/lorawan-stack/issues/2353).

> but it still break very often

You mean the events stream stop working?

wasn-eu, 2020-05-06:
yes the stream stops and i have to click restart stream.

will try to write it down what is needed.

wasn-eu, 2020-05-06:
wrote what i have done: https://github.com/TheThingsNetwork/lorawan-stack/issues/2353#issuecomment-624583182

bafonins, 2020-05-12:
> yes the stream stops and i have to click restart stream.

I will take a look if there are any in the way handle stream connections.

laurensslats, 2020-06-16:
This issue still occurs. When the error occurs, it is not possible to click "Restart stream" and the page has to be refreshed which removes all the cashed data.

[image]

[image]

closed 2020-06-20 by bafonins