TheThingsNetwork/lorawan-stack #2392: Console ignores errors in tentity title section

bafonins, 2020-04-22:
#### Summary

The Console ignores failed requests when fetching data for the entity title section. This results in an infinite loading indicator.

[image]
[image]
[image]

#### Steps to Reproduce

1. Simulate api-keys requests for the entity title section
2. No error UI

#### What do you see now?

Infinite loading indicator

#### What do you want to see instead?

Error UI, or skipped entity count for failed requests

#### Environment

v3.7.1

#### How do you propose to implement this?

Selects errors and change the way the loading state is determined for the entity title section

#### Can you do this yourself and submit a Pull Request?

yes

bafonins, 2020-05-27:
@kschiffer any suggestions on how it should look?

kschiffer, 2020-05-29:
I think the section should try to show all values that have been fetched successfully, for unretrievable values, it should show e.g. `API Keys n/a` using the `warning` icon instead of the entity icon, everything rendered in `$tc-error`.

Hope that's clear enough, otherwise I can do a mock-up.

bafonins, 2020-06-21:
Blocked on https://github.com/TheThingsNetwork/lorawan-stack/issues/2740. I want to refactor the whole thing altogether

closed 2020-09-08 by bafonins