TheThingsNetwork/lorawan-stack #2403: Fix AppSKey handling in AS

bafonins, 2020-04-23:
#### Summary

AS allows resetting AppSKey but throws an error when querying for `session`.

#### Steps to Reproduce

1. Query an abp device from AS with `app_s_key` set
2. Reset `app_s_key` to an empty string (note, empty object and `null` will work as well)
3. Query the device again, see the error

```
$ curl -H "Authorization: Bearer $access_token" "http://localhost:8080/api/v3/as/applications/test-app/devices/test-abp?field_mask=session" | jq
{
  "ids": {
    "device_id": "test-abp",
    "application_ids": {
      "application_id": "test-app"
    },
    "dev_eui": "8971238978978978"
  },
  "created_at": "2020-04-23T07:03:13.650309Z",
  "updated_at": "2020-04-23T07:03:13.650309Z",
  "session": {
    "dev_addr": "AD897897",
    "keys": {
      "app_s_key": {
        "key": "38971237123812871189731289731289"
      }
    },
    "started_at": "0001-01-01T00:00:00Z"
  }
}
$ curl -X PUT -H "Authorization: Bearer $access_token" --data '{
   "end_device":{                                        
      "session":{
         "keys":{
            "app_s_key":{
               "key": ""  
            }
         }
      }
   },
   "field_mask":{
      "paths":[
         "session.keys.app_s_key.key"
      ]
   }
}' http://localhost:8080/api/v3/as/applications/test-app/devices/test-abp | jq
{
  "ids": {
    "device_id": "test-abp",
    "application_ids": {
      "application_id": "test-app"
    },
    "dev_eui": "8971238978978978"
  },
  "created_at": "2020-04-23T07:03:13.650309Z",
  "updated_at": "2020-04-23T07:12:14.445113Z",
  "session": {
    "dev_addr": "00000000",
    "keys": {
      "app_s_key": {}
    },
    "started_at": "0001-01-01T00:00:00Z"
  }
}
$ curl -H "Authorization: Bearer $access_token" "http://localhost:8080/api/v3/as/applications/test-app/devices/test-abp?field_mask=session" | jq
{
  "code": 3,
  "message": "error:pkg/crypto/cryptoutil:invalid_length (invalid slice length)",
  "details": [
    {
      "@type": "type.googleapis.com/ttn.lorawan.v3.ErrorDetails",
      "namespace": "pkg/crypto/cryptoutil",
      "name": "invalid_length",
      "message_format": "invalid slice length",
      "correlation_id": "87f07361d59847ec90da5bdbde998a18",
      "code": 3
    }
  ]
}
```

#### What do you see now?

Error on get request

#### What do you want to see instead?

If we allow resetting `app_s_key` then there shouldnt be an error on gets.
If we dont allow resetting `app_s_key` the error should be returned on update.

#### Environment

v3.7.1

#### How do you propose to implement this?

Most probably some validation issues

#### Can you do this yourself and submit a Pull Request?

@rvolosatovs

kschiffer, 2020-05-13:
[This Sentry log](https://sentry.io/organizations/the-things-industries/issues/1661193139/?project=2682566&referrer=alert_email) suggests that this issue might still be unresolved (on 3.8.1) or at least that there is still a very similar issue. Can someone verify?

```json
{"type":"GET_DEVICE_FAILURE","payload":{"message":"error:pkg/crypto/cryptoutil:invalid_length (invalid slice length)","code":3,"details":"[Array]"},"error":true}
```
Happening on an ABP device.

rvolosatovs, 2020-05-14:
This is the error I see: 
[image]

#2481 made it impossible for devices to reach the invalid state, but if we already have devices in the registry with this invalid state, we should indeed add handling for that.

johanstokking, 2020-06-09:
What is the status here?

rvolosatovs, 2020-06-10:
@kschiffer can you send a sentry link to the actual error or give steps to reproduce the issue?

kschiffer, 2020-06-10:
I can't give you anything more than the sentry link above unfortunately. After #2527 we can hopefully make more sence of the error, if it occurs again.

kschiffer, 2020-06-11:
Sorry, that was indeed the wrong sentry issue and I can't find the relevant one anymore. I suggest to close this until we bump into this again.

PR #2481 (2020-05-07): Fix key handling in device registries

#### Summary

Closes #2403 

#### Changes

- Make key handling in device registries consistent, deleting keys on `SET` is now disallowed.
	- For Join Server it does not make sense to have a device with no keys.
	- For Application Server there might be a use case to delete `AppSKey` if the user wants to migrate to an external AS, but we can add support for this later - currently deleting keys is broken anyway, so this makes the issue apparent instead of silently letting the device reach an invalid state in the registry.
- Remove unused/disallowed fieldpath [... cut]
closed 2020-06-11 by kschiffer