TheThingsNetwork/lorawan-stack #2404: Improve UX for ByteInputs without fixed value length

kschiffer, 2020-04-23:
#### Summary
Our `ByteInput` component is not really working well for getting byte values without length restriction.

#### Why do we need this?
We need to get byte values without length restriction in the schedule downlink secrtion.

#### What is already there? What do you see now?
ByteInput component enabling inputs without length restriction.

#### What is missing? What do you want to see?
A better behavior of such input: don't show the masked input dots.

#### Environment
v3.7.1

#### How do you propose to implement this?
I think it's best to refactor the byte input into its own component. We currently have it as variation of the normal `<Input />` component, which makes it hard to maintain and extend and it also arbitrarily mixes prop types between the two.

#### Can you do this yourself and submit a Pull Request?
@pgalic96

bafonins, 2020-05-11:
@pgalic96 is this closed?

pgalic96, 2020-05-11:
yes.

PR #2450 (2020-05-08): Improve byte input UX

#### Summary

Closes #2404 

#### Changes

- Fix the issue with showing placeholder chars when input is unbounded
- Fix the issue with pasting values when input is unbounded

#### Notes for Reviewers

cc @kschiffer . I have looked into moving `ByteInput` to a separate component, but I am not sure this is the best idea. It would include a lot of replication since there is still a lot of similarities with the `Input` component. I have managed to fix this issue by combination of `showMask` and `guide` flags in the input. I think then most of the issues with this input are s [... cut]
closed 2020-05-11 by pgalic96