TheThingsNetwork/lorawan-stack #2406: Don't show gateway as disconnected if they are connected in a different region

laurensslats, 2020-04-23:
#### Summary
Gateways that are connected to a different region than the user is currently looking at seem offline, though they are online.

...

#### Why do we need this?
It seems like false information. 
...

#### What is already there? What do you see now?
You do see: `Other Cluster` in the specific gateway view, but not in the list with all gateway. See printscreens below.

[image]

[image]

...

#### What is missing? What do you want to see?
Change `Other Cluster` to `Other Region`. Show this in the gateway list and in the gateway page. See printscreens below:

[image]

[image]

...

#### How do you propose to implement this?
Easy fix.

...

laurensslats, 2020-04-23:
I'd say **Other Region** is clearer than **Other Cluster**. Let me know if you think otherwise.

bafonins, 2020-04-23:
> The gateway is connected to `nam1` and I'm looking at the `eu1` console.

We show the `Other Cluster` indicator only if `gateway_server_address` configured for the gateway is not the same as the one in the current deployment. To be more specific hostname of `eu1` is `tti.eu1.cloud.thethings.industries` while `gateway_server_address` is configured to `tti.nam1.cloud.thethings.industries`

> Show this in the gateway list and in the gateway page. 

Indeed, we should do that.

@htdvisser what about changing `Other Cluster` to `Other Region`? Im fine with this. Here is the initial discussion regarding this: https://github.com/TheThingsNetwork/lorawan-stack/issues/243#issuecomment-489601891

htdvisser, 2020-04-23:
The correct term is "Cluster". We can have `eu1` and `eu2` clusters that are the same region, but not the same cluster.

closed 2020-04-30 by bafonins