TheThingsNetwork/lorawan-stack #2410: Update join flow example in the events guide

bafonins, 2020-04-23:
#### Summary

We should update the join flow example in the docs.

#### Why do we need this ?

To be consistent with the events used in the join flow.

#### What is already there? What do you see now?

https://github.com/TheThingsNetwork/lorawan-stack/blob/develop/doc/content/guides/events/_index.md#L46-L211

#### What is missing? What do you want to see?

Updated snapshot of events happening during the join flow

#### How do you propose to document this?

The same way, but with updated events snapshot

#### Can you do this yourself and submit a Pull Request?

yes

bafonins, 2020-04-23:
cc @benolayinka

benolayinka, 2020-06-16:
@bafonins it will take me a lot longer than you to do this, can you do it?

bafonins, 2020-10-02:
Here are the events dispatched when a device join the network:
```js
{
    "name": "ns.up.join.receive",
    "time": "2020-10-02T08:28:16.902963518Z",
    "identifiers": [
      {
        "device_ids": {
          "device_id": "uno-otaa",
          "application_ids": {
            "application_id": "example-application"
          },
          "dev_eui": "0004A30B001BDAED",
          "join_eui": "58A0CB0001500001"
        }
      }
    ],
    "data": {
      "@type": "type.googleapis.com/ttn.lorawan.v3.UplinkMessage",
      "raw_payload": "AAEAUAEAy6BY7dobAAujBACzaHMSWIU=",
      "payload": {
        "m_hdr": {},
        "mic": "cxJYhQ==",
        "join_request_payload": {
          "join_eui": "58A0CB0001500001",
          "dev_eui": "0004A30B001BDAED",
          "dev_nonce": "68B3"
        }
      },
      "settings": {
        "data_rate": {
          "lora": {
            "bandwidth": 125000,
            "spreading_factor": 7
          }
        },
        "data_rate_index": 5,
        "coding_rate": "4/5",
        "frequency": "868300000",
        "timestamp": 2292083595
      },
      "rx_metadata": [
        {
          "gateway_ids": {
            "gateway_id": "ttkg"
          },
          "time": "2020-10-02T08:28:19Z",
          "timestamp": 2292083595,
          "rssi": -58,
          "channel_rssi": -58,
          "snr": 10.5,
          "uplink_token": "ChAKDgoMdHRrZy1ib2dkYW5zEIvX+cQIGgwIoMnb+wUQlonR3QIg+K3Q1tqnCQ=="
        }
      ],
      "received_at": "2020-10-02T08:28:16.799062784Z",
      "correlation_ids": [
        "gs:conn:01EKJZRWEK1APM8H6C4TAEGPH2",
        "gs:up:host:01EKJZRWEX2TQ5GCRNVPJZDQT7",
        "gs:uplink:01EKM6T4YXP8XKCSRVDJMKZXF3",
        "ns:uplink:01EKM6T50ZCT9T121JQK7AJK0W",
        "rpc:/ttn.lorawan.v3.GsNs/HandleUplink:01EKM6T50YXZP05609QMJH17GV"
      ],
      "device_channel_index": 1
    },
    "correlation_ids": [
      "gs:conn:01EKJZRWEK1APM8H6C4TAEGPH2",
      "gs:up:host:01EKJZRWEX2TQ5GCRNVPJZDQT7",
      "gs:uplink:01EKM6T4YXP8XKCSRVDJMKZXF3",
      "ns:uplink:01EKM6T50ZCT9T121JQK7AJK0W",
      "rpc:/ttn.lorawan.v3.GsNs/HandleUplink:01EKM6T50YXZP05609QMJH17GV"
[... 448 lines cut ...]
      "@type": "type.googleapis.com/ttn.lorawan.v3.ScheduleDownlinkResponse",
      "delay": "2.117555039s"
    },
    "correlation_ids": [
      "gs:conn:01EKJZRWEK1APM8H6C4TAEGPH2",
      "gs:up:host:01EKJZRWEX2TQ5GCRNVPJZDQT7",
      "gs:uplink:01EKM6T4YXP8XKCSRVDJMKZXF3",
      "ns:downlink:01EKM6T7RTXXQ3K8ZB58JBR82X",
      "ns:uplink:01EKM6T50ZCT9T121JQK7AJK0W",
      "rpc:/ttn.lorawan.v3.GsNs/HandleUplink:01EKM6T50YXZP05609QMJH17GV"
    ],
    "origin": "ip-10-20-6-248.eu-west-1.compute.internal",
    "context": {
      "tenant-id": "CgN0dGk="
    },
    "visibility": {
      "rights": ["RIGHT_APPLICATION_TRAFFIC_READ"]
    },
    "unique_id": "01EKM6T7S3KKWHYGM06B3T00HP"
  }
```

The list is much longer that what we currently have. Do we really want to keep this in the documentation? Especially taking into account that this flow will change in the future (by removing/adding events). Do we really want to keep this synced with the docs?

@benolayinka @laurensslats

johanstokking, 2020-11-12:
@benolayinka please re-open if this is still relevant.

closed 2020-11-12 by johanstokking