TheThingsNetwork/lorawan-stack #2412: LoRaCloud DAS users should be able to select their own server URL

Oliv4945, 2020-04-23:
#### Summary
LoRaCloud users should be able to select their own DAS server URL as it will not always be `https://das.loracloud.com`

#### How do you propose to implement this?
Add a `das-server-url` option taking and URL input

#### Can you do this yourself and submit a Pull Request?
Seems difficult, but maybe if I am helped with the steps to do it

johanstokking, 2020-04-24:
@Oliv4945 what is the scope of this; per-integration? Not per device I hope?

johanstokking, 2020-04-24:
Adding `compat/api` for the users to customize the URL.

@adriansmares I'm quite sure the PR needs `bump/minor` as an old CLI or Console may mess up the new settings

adriansmares, 2020-04-24:
Application packages have their configuration stored on a per-device basis (due to the nature of the FPort identifiers), so this would be indeed per-device. Nonetheless, the configuration is easily loaded through CLI so mass-provisioning this is fine.

Regarding the CLI question - actually settings are loaded as JSON (we use `proto.Struct`), so old CLI's won't break this, and we can ensure that missing values default to the preconfigured default.

johanstokking, 2020-04-24:
OK great!

Oliv4945, 2020-04-28:
@johanstokking @adriansmares As discussed during TTN conference application packages as per-devices, but it would be great to be able to have a configuration per application like the payload decoder.

PR #2539 (2020-05-15): Support application wide application packages associations

#### Summary

Closes #2412
Closes #2473
Closes #2474

#### Changes

- Add application packages default associations. These are the application-wide equivalent of normal application packages associations. They allow users to setup an application package for the whole application instead of an per-device basis. 
- Application packages may now handle _unrelated_ traffic (traffic on a different FPort than the one they've been associated with). This allows application packages to schedule downlinks even when their FPort hasn't been used.
- [... cut]
closed 2020-05-15 by adriansmares