TheThingsNetwork/lorawan-stack #2415: doc: Hidden nav item css breakpoints

benolayinka, 2020-04-24:
#### Summary

After doc restructuring #2368 , nav items don't fit on small screens 
[image]

#### Steps to Reproduce

1. Pull the master branch and run the docs server.
2. Resize your window to approx 1050px wide

#### What do you see now?

[image]

#### What do you want to see instead?

Maybe burger menu? 

[image]

#### Environment

Chrome 80

#### How do you propose to implement this?

Css breakpoints?

#### Can you do this yourself and submit a Pull Request?

Not cleanly. @kschiffer @htdvisser @pgalic96 ideas?

htdvisser, 2020-04-24:
Changing the breakpoint is easy (see below), but I don't think it improves the user experience. Maybe we should reconsider whether we really want the "Github", "Forum" and "Get The Things Stack" links in the top bar, or if those can actually be removed. The "Github" and "Forum" links are also in the footer, and we also have a huge "hero" on the homepage with a "Get The Things Stack" link.

This is how you change the breakpoint (default is `960px + (2 * $gap)`):

```diff
diff --git a/doc/themes/the-things-stack/assets/css/theme.scss b/doc/themes/the-things-stack/assets/css/theme.scss
index 5b146d66f..c5a984c5f 100644
--- a/doc/themes/the-things-stack/assets/css/theme.scss
+++ b/doc/themes/the-things-stack/assets/css/theme.scss
@@ -56,6 +56,8 @@ $table-cell-border: 1px solid $white-ter;
 $table-cell-border-width: 1px 1px 1px;
 $table-cell-padding: 1.2em 0.75em;
 
+$desktop: 1024px + (2 * $gap);
+
 @import "bulma/sass/utilities/functions.sass";
 @import "bulma/sass/utilities/derived-variables.sass";
 @import "bulma/sass/utilities/animations.sass";

```

benolayinka, 2020-04-24:
> Changing the breakpoint is easy (see below), but I don't think it improves the user experience. Maybe we should reconsider whether we really want the "Github", "Forum" and "Get The Things Stack" [...]

It's tough. I think Github and Forum are important links for top bar nav - I always click these when I visit docs.

Get The Things Stack could become Download (i guess it was before since that's the slug it redirects to). I prefer Download, but maybe there were reasons for changing this CTA.

kschiffer, 2020-04-25:
> It's tough. I think Github and Forum are important links for top bar nav - I always click these when I visit docs.

I agree. These should stay there IMO. We might rather get rid of the version tag next to the logo, since we'll be getting rid of versioned docs soon (if I understood correctly) and we can also display that in the footer.

> Get The Things Stack could become Download

Agreed.

benolayinka, 2020-05-13:
Enterprise docs don't have the Download button and everything fits nicely - maybe we (re)move that button?

[image]

benolayinka, 2020-09-23:
Closed by https://github.com/TheThingsIndustries/lorawan-stack-docs/pull/2

closed 2020-09-23 by benolayinka