TheThingsNetwork/lorawan-stack #2421: Document available frequency plans

neoaggelos, 2020-04-27:
#### Summary

Add a "Reference" section with a list of the available frequency plans and their descriptions. Possibly also add a note for the ones that are irrelevant when configuring gateways (ref #2140) 

#### Why do we need this ?

Helping v2 users migration. v2 users only had a very simple option to choose from with v2, but now they are presented with a wider array of frequency plans for gateways, without any clear indication for which they should choose.

Also, on the v3 we require choosing a frequency plan, and LoraWAN PHY and MAC versions for the devices as well. This may alienate them and make migrating harder, since none of that was needed on v2
 
#### What is already there? What do you see now?

A dropdown list for choosing frequency plans (gateways + devices) and lorawan versions (for devices) 

A repository for the available frequency plans https://github.com/TheThingsNetwork/lorawan-frequency-plans

#### What is missing? What do you want to see?

A documentation page for the available frequency plans and a clear description of each. This could be possibly generated dynamically from https://github.com/TheThingsNetwork/lorawan-frequency-plans/blob/master/frequency-plans.yml with some Hugo magic

Also, the frequency plans under the same radio frequencies are not relevant for gateways. So, it would be helpful to let the users know about that.

#### How do you propose to document this?

Create a page in the reference section of our documentation for these. Also add liniks from the stack to that page (ref #2397)

#### Can you do this yourself and submit a Pull Request?

assigning to @benolayinka , but i can help with that

benolayinka, 2020-04-28:
@htdvisser also created an issue for this

https://github.com/TheThingsNetwork/lorawan-frequency-plans/issues/4

johanstokking, 2020-04-29:
I think that this is changing so little that we don't really need the Hugo magic. Let's not over-engineer this for now.

benolayinka, 2020-04-30:
The most confusing for me was the separate TTN band. I'm still not sure what it means?

Beyond that, maybe a note in the console-'
"Choose the frequency plan that matches your geographical area"

Are devices physically limited to a frequency range within an area? Would this be printed on the device?

[image]

johanstokking, 2020-04-30:
It's not a TTN band but a frequency plan.

There are two layers: bands and frequency plans. A frequency plan is for a specific band. A band is defined by the LoRaWAN Regional Parameters, and which band(s) the device supports (and is certified for) can typically be found in the data sheet of the device. A device can only be configured to use one band at a time.

So one must select a frequency plan for the band that the device supports.

johanstokking, 2020-04-30:
We can definitely clarify this indeed.

benolayinka, 2020-04-30:
> It's not a TTN band but a frequency plan.
>
> There are two layers: bands and frequency plans. A frequency plan is for a specific band. A band is defined by the LoRaWAN Regional Parameters, and which [...]

Even in the LoRa spec I'm having trouble finding much information about Frequency Plans. It looks like it has to do with beaconing or listening frequency hopping?

If the device supports a band, what's the difference between picking two frequency plans within that band in the console? Does the network server then configure the device to support that frequency plan as part of activation?

johanstokking, 2020-04-30:
> Even in the LoRa spec I'm having trouble finding much information about Frequency Plans. It looks like it has to do with beaconing or listening frequency hopping?

Frequency plans are in LoRaWAN called `CFList` (channel frequency list) or `ChMask` (channel mask).

Bands are defined by Regional Parameters and follow regulations. Frequency plans are a list of frequencies that comply with regulations for a given band.

There are two types of bands; those with dynamic channels (i.e. EU868) that use the `CFList` to communicate channels by frequency, and fixed channels (i.e. US902, AU915) that use `ChMask` to enable/disable channels by number (bitmasking).

Next to frequencies, the frequency plan also contains data rates, and that is also where differences are.

> If the device supports a band, what's the difference between picking two frequency plans within that band in the console? Does the network server then configure the device to support that frequency plan as part of activation?

Yes, `CFList` and `ChMask` are part of the join-accept message.

benolayinka, 2020-05-04:
> Frequency plans are in LoRaWAN called `CFList` (channel frequency list) or `ChMask` (channel mask).
>
> Bands are defined by Regional Parameters and follow regulations. Frequency plans are a list of [...]

Thanks. I'm going to update the glossary with these terms:
- Frequency Band: For LoRaWAN, a Frequency Band is a range of frequencies divided either in to dynamic channels (i.e. EU868) or fixed channels (i.e. US902, AU915). The LoRaWAN Regional Parameters specify which Frequency Bands are supported by LoRaWAN in a geographical area. Within a Frequency Band, there can be multiple complying Frequency Plans. Devices typically support one or more Frequency Bands in their hardware, and are configured for a particular Frequency Plan as part of activation.
- Frequency Plan: A Frequency Plan complies with the regulations specified in the LoRaWAN Regional Parameters for a particular Frequency Band. A Frequency Plan can additionally support different data rates within that Frequency Band. Devices support any Frequency Plan within a supported Frequency Band, and are configured for a particular Frequency Plan as part of activation.
- Dynamic Channel: A Frequency Band which uses Dynamic Channels (i.e. EU868) uses a Channel Frequency List (CFList) to communicate channels by frequency.
- Fixed Channel: A Frequency Band which uses Fixed Channels (i.e. US902, AU915) uses a Channel Mask (ChMask) to enable and disable channels.

johanstokking, 2020-05-05:
Looks good. Two things;

- I would not call it Frequency Band but "Band"
- I would add to Frequency Plan that it is also sometimes known as a Channel Plan

benolayinka, 2020-06-09:
Closed by #2562

PR #2562 (2020-06-09): Add frequency plans to docs

#### Summary

Closes #2421 

#### Changes

- Add frequency plans reference to docs
- ~~Copypasta'd frequency plans from https://github.com/TheThingsNetwork/lorawan-frequency-plans/blob/master/frequency-plans.yml to /data~~
- Css for frequency plan table
- Add step to download frequency-plans.yml in `./mage docs:deps`

#### Notes for Reviewers

**Discussion:** What's the best way to link to the yaml data? I copied https://github.com/TheThingsNetwork/lorawan-frequency-plans/blob/master/frequency-plans.yml to the data section of the docs manually, which is not a solut [... cut]
closed 2020-06-09 by benolayinka