TheThingsNetwork/lorawan-stack #2424: Rename PHY and MAC in console

benolayinka, 2020-04-28:
#### Summary

It's unclear for newbies what MAC and PHY are in the console. The LoRaWAN terms for these are (i think)
MAC: LoRaWAN specification or version
PHY: Regional parameters

#### Why do we need this ?

To make it easy to add devices in the v3 console (these terms are also unfamiliar for those upgrading from v2, where they were not required)

#### What is already there? What do you see now?

[image]

#### What is missing? What do you want to see?

MAC and PHY replaced with terms used by LoRa Alliance / other device documentation

[image]

#### How do you propose to document this?

This is a change to documentation

#### Can you do this yourself and submit a Pull Request?

No, @kschiffer @pgalic96 ?

@htdvisser @johanstokking can you double check my LoRa terminology

johanstokking, 2020-04-28:
I would go with:

- **LoRaWAN version**
- **Regional Parameters version**

pgalic96, 2020-05-11:
Closed by #2451

PR #2451 (2020-05-08): Rename MAC and PHY fields in console

#### Summary

Closes #2424 

#### Changes

- Change `MAC version` to `LoRaWAN version` and `PHY version` to `Regional Parameters version` 

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API, storage, configuration and CLI, according to the compatibility commitments in `README.md`. The target branch is set to `master` if the changes are fully compatible with existing API, database, configuration and CLI.
- [x] Testing: The changes are co [... cut]
closed 2020-05-11 by pgalic96