TheThingsNetwork/lorawan-stack #2431: Support BeaconTimingReq

rvolosatovs, 2020-04-28:
#### Summary

Support BeaconTimingReq

#### Why do we need this?

To support Class B in pre-1.0.3 devices

#### What is already there? What do you see now?

Class B support for 1.0.3+ devices

#### What is missing? What do you want to see?

`BeaconTimingReq` support

#### How do you propose to implement this?

Just like all other MAC commands

#### Can you do this yourself and submit a Pull Request?

yes

johanstokking, 2021-12-30:
I don't think we need this. For the record, this would only be needed for end devices using LoRaWAN 1.0.2 or lower that use class B. I'm leaving it open for users to comment that they need it based on their use cases.

KrishnaIyer, 2025-04-30:
We haven't received any updates here for years now. Let's close this.

closed 2025-04-30 by KrishnaIyer