TheThingsNetwork/lorawan-stack #2434: Resetting frame counts for end devices in the Console

kschiffer, 2020-04-29:
#### Summary
Users should be able to reset the end device frame counters.

#### Why do we need this?
Close feature gap with the v2 console

#### What is already there? What do you see now?
I guess the backend already supports this, but I need some info on that (cc @rvolosatovs)

#### What is missing? What do you want to see?
A way to trigger the frame counter reset in the console.

#### Environment
v3.7.1

#### How do you propose to implement this?
I'd say this could be a simple button inside of the Network Layer tab in the end device settings.

#### Can you do this yourself and submit a Pull Request?
Yes, but I need some hint as to what API to use for this.

rvolosatovs, 2020-04-29:
Are we talking about the uplink frame counters here?
If so, resetting uplink frame counters must always include MAC state reset as well, which is more tricky and requires custom logic in Network Server. In any case Network Server automatically determines and handles uplink frame counter resets for devices, for which `resets_f_cnt == true`, I don't think we need an explicit way to reset uplink frame counters from the console. cc @johanstokking

johanstokking, 2020-04-29:
I think having a button "reset MAC state" with an info message that this also resets frame counters, is a good thing.

This is for devices that are not LoRaWAN compliant and loose their internal state on a power cycle. So yes, it is more than frame counters; it's resetting the whole thing, via the Console.

rvolosatovs, 2020-04-29:
> This is for devices that are not LoRaWAN compliant and loose their internal state on a power cycle. So yes, it is more than frame counters; it's resetting the whole thing, via the Console.

The devices, which are not LoRaWAN compliant in this case will send an uplink with frame counter lower than the previous one - Network Server will notice that and reset the MAC state without any user input.
See https://github.com/TheThingsNetwork/lorawan-stack/blob/0acff7cd288d114b43f6b178667cdcfb6a11c4ac/pkg/networkserver/grpc_gsns.go#L346-L416

While I agree that this is nice to have, this is definitely not a priority in my opinion and is a "goldplating" issue, since this is already handled automatically by the Network Server.

johanstokking, 2020-04-29:
> The devices, which are not LoRaWAN compliant in this case will send an uplink with frame counter lower than the previous one - Network Server will notice that and reset the MAC state without any user input.

Only when `resets_f_cnt` is set. The use case here is that this is a manual action, because the owner knows that this happened. The goal here is to avoid people setting this too often, because that opens the door for any spoofer to reset the MAC state all the time.

rvolosatovs, 2020-04-29:
Blocked by https://github.com/TheThingsNetwork/lorawan-stack/issues/2446

rvolosatovs, 2020-04-29:
Once https://github.com/TheThingsNetwork/lorawan-stack/issues/2446 is implemented, console will have to just set `session.last_f_cnt_up` to `0`

johanstokking, 2020-10-27:
@rvolosatovs I don't think we settled on not having a button. I think we should reset automatically with `resets_f_cnt` is set, but we _also_ need a reset button to force-reset (even if `resets_f_cnt` is not set).

Please look if #2446 is actually blocking this issue.

rvolosatovs, 2020-10-27:
It is indeed, see https://github.com/TheThingsNetwork/lorawan-stack/issues/2446#issuecomment-717351969
We settled on having a button, which does this: https://github.com/TheThingsNetwork/lorawan-stack/issues/2434#issuecomment-621117680

johanstokking, 2020-10-27:
OK, so. the labels are correct then.

closed 2021-11-18 by bafonins