TheThingsNetwork/lorawan-stack #2435: Provide web UI and backend message language translations

kschiffer, 2020-04-29:
#### Summary
Our web UI is set up for internationalization. We should start to make use of that by translating the frontend.

#### Why do we need this?
Increase the accessibility of our services for non-English speaking people.

#### What is already there? What do you see now?
Locale files that get automatically generated and updated as we go. See `pkg/webui/locales/en.json`.

#### What is missing? What do you want to see?
Duplicating and translating the English locale file of the Web UI.
Next to this, we also have the backend messages which can and should be translated as well.

Foremost for (top of my head list):
- Chinese
- Japanese
- Russian
- Spanish
- French
- German

#### How do you propose to implement this?
We would need volunteers (or professional translating services?) with a brief instruction of the locale files add locale files in their own languages.

#### Can you do this yourself and submit a Pull Request?
I can do the German translation but I don't think that is particularly a priority. More important is Japanese and Chinese.

Let's make a plan for how to proceed with this.

laurensslats, 2020-05-12:
I would suggest starting with Japanese.

@kschiffer How can someone submit the translations, do we have a clear format on how to do so?

rish-c, 2020-05-12:
There are 2 parts that needs to be translated - 
* Component Description
* Display Text
For any new language to be added to the console, we need the translated version of (1) and (2) as `xx.json`. 
https://github.com/TheThingsNetwork/lorawan-stack/blob/develop/pkg/webui/locales/xx.json

kschiffer, 2020-05-13:
> There are 2 parts that needs to be translated -
> 
> * Component Description

What do you mean?

> * Display Text
>   For any new language to be added to the console, we need the translated version of (1) and (2) as `xx.json`.
>   https://github.com/TheThingsNetwork/lorawan-stack/blob/develop/pkg/webui/locales/xx.json

Disregard `xx.json`, this is only for development purposes. No need to do anything with that.

rish-c, 2020-05-13:
For example - 
Component Description - "components.data-sheet.index.noData"
English Text - "No data available"

So we need the translated version for all the text which appears under Component description and the corresponding text to display; all of this in a file such as `cn.json`, `jp.json`.

kschiffer, 2020-05-13:
No that's not true. The component column does not need to be translated, only the corresponding text. The component column is basically an identifier for the message, which hence needs to stay the same.

htdvisser, 2020-08-24:
Moving to Backlog until we have translators who can maintain this.

wienke, 2020-12-04:
So is it time now to start hiring translators? I mean we can already start with:

- German - Kevin
- French - Pierre/Ines
- Dutch - Laurens
- Italian - Ines

And we can do others like Spanish, Portuguese, Chinese and Japanese through fiver.com

So is there anything holding us back from starting this process?

kschiffer, 2021-03-04:
I suggest the following to make this actionable:

*   Adjust the `mg js:clean` command to only delete `en.json` and `xx.json`
*   Extend the user registry with a `LanguagePreference` field
*   Revisit our build process around locales
    *   We should avoid making the default language preference influence the build process
*   Extend the profile settings in the Account App with a language chooser
*   Make sure the OS language is used by default (which should be passed to the browser via `navigator.language`)
*   Add logic to fallback to english for any non-english message that hasn't been included yet. This is currently already done but triggers a console error.
*   Add syntax check for the ICU syntax for all locale files, as part of CI
*   Add an acknowledgements page to the docs which lists the volunteers that helped translating. This page should then be linked in the Console.
*   Onboard community members that want to translate messages with a guide about contributing locale files
    *   This should include a simple way to retrieve untranslated messages of a given locale
*   Based on community support, revisit locales regularly to ensure that message gaps are resolved timely

johanstokking, 2021-04-12:
@kschiffer please assign someone who can pick this up. @asmulko maybe?

johanstokking, 2021-07-13:
Removing this from the planning for now.

kschiffer, 2021-07-27:
Currently working on a slimmed-down preliminary solution for this that will enable us to add the Japanese locale already.

johanstokking, 2021-07-27:
> Currently working on a slimmed-down preliminary solution for this that will enable us to add the Japanese locale already.

Nice! 🙏

kschiffer, 2021-07-30:
With #4465 merged, we will have a starting point for i18n that also includes the Japanese locale.

Here's how I think we should proceed from there, we should:

*   change the JSON marshaling within the `i18n` package to accept (and write) locale files per language in the same simple key->value format that we use for the frontend
*   Change message loading logic in a way that regions can be accepted, e.g. `en_GB`. This means that regional locales are imported (if available) and merged with the base language (`en`)
*   Instead of handling backend translations in the frontend via the message ID, the backend should already provide translated text based on the `Accept-Language` header.
*   use the `Accept-Language` header also to render the `lang` attribute in the HTML template of the web UIs
*   use the user registry to store language preference (currently uses `LocalStorage`)
*   optionally provide a new config option to limit the available languages (though not sure if this is really necessary)
*   move all locale files to a top-level `locales` folder
*   use a translation management app like Weblate that can
    *   Enable crowd-sourced translations
    *   Store WIP translation work until a threshold of translated messages is reached
    *   Connect to the Github repo and maintain a `i18n` branch that we can then manually merge into master in regular intervals

cc @htdvisser @johanstokking

NicolasMrad, 2022-06-27:
@kschiffer can this be closed?

kschiffer, 2022-06-29:
Likely, but let's discuss this one as well in the stack meeting before.

htdvisser, 2022-07-01:
For the record, last year I worked on the backend side of message translations (https://github.com/TheThingsNetwork/lorawan-stack/tree/feature/2435-backend-i18n).

[image]

However, that branch was never finished, because there wasn't really any demand for this, so if we decide to close this issue, then that branch can also be deleted.

NicolasMrad, 2022-07-05:
let's keep the branch, but closing the issue for now

closed 2022-07-05 by NicolasMrad