TheThingsNetwork/lorawan-stack #2436: Determine end device statuses

kschiffer, 2020-04-29:
#### Summary
The console should provide a status indication of an end device.

#### Why do we need this?
- To facilitate and speed up situational awareness for end devices of an application
- To close feature gap with the v2 console

#### What is already there? What do you see now?
- End device list
- End device event stream

#### What is missing? What do you want to see?
Some logic to inform a discrete end device status and displaying that in the end device overview and end device list

#### How do you propose to implement this?
The question is what we want to display as a status and how we want to determine that status. It is not trivial to generate an end device status from the data that it holds. In v2, we determined a certain time span of not sending a message as an indication of the device being down. However, this must not be the case. Devices can very well go without sending or receiving data for a long time depending on the usage scenario.

Let's discuss what would be a good way to provide end device statuses.

#### Can you do this yourself and submit a Pull Request?
After we have found a strategy, yes.

htdvisser, 2020-04-29:
Issues can only be in Next Up if they are ready to be picked up. This issue still needs discussion.

htdvisser, 2020-04-29:
I don't think there is such a thing as "device status". Best thing we can do is display a "time ago" and perhaps a signal quality for the last uplink

laurensslats, 2020-05-13:
When we know the last uplink in the application, let's replace `Linked` with `Last seen` in the application view (similar to the gateway overview page).

[image]

johanstokking, 2021-05-18:
@kschiffer to which extent does #4132 (partly) close this issue?

kschiffer, 2021-05-18:
Yes I think this is practically solved by #4132, since we cannot really properly determine any "device status" and the best we can do is forwarding this consideration to the user by providing them with the "Last seen".

I'll set #4132 to close this issue.

PR #4132 (2021-05-20): Display last seen information of end device in the end device table
#### Summary
This quickfix displays the live last seen info (last received uplink) of devices in the device tables.

[image]

Closes #2436
Closes #3832

#### Changes
- Update store logic to fetch mac state of devices when fetching the list of devices
- Update device table to show the "last seen" info, based on the mac state
- Update the JS SDK to enable whitelisting stack components for device get requests

#### Testing

Manual testing

#### Notes for Reviewers
This will make an additional request per device of the device list [... cut]
closed 2021-05-20 by kschiffer