TheThingsNetwork/lorawan-stack #2438: Allow setting frame counter widths of end devices in the Console

kschiffer, 2020-04-29:
#### Summary
The console should allow setting the frame counter widths that an end device uses.

#### Why do we need this?
- It's a basic setting
- To close v2/v3 console feature gap

#### What is already there? What do you see now?
No such setting.

#### What is missing? What do you want to see?
A simple setting to switch between 16 and 32 bit frame counter widths.

#### How do you propose to implement this?
Given the respective API, this should be trivial. The setting could be part of the Network Layer tab in the end device settings.

#### Can you do this yourself and submit a Pull Request?
Yes.

bafonins, 2020-04-29:
@rvolosatovs is the `supports_32_bit_f_cnt` field the one needed for this feature?

rvolosatovs, 2020-04-29:
Yes

pgalic96, 2020-09-24:
@kschiffer @bafonins Should this setting be part of the add end device wizard, or should we keep it in the general settings only to keep the wizard simple (I assume the default value is `false` then).

pgalic96, 2020-09-30:
@rvolosatovs looping you in on the question regarding `supports_32_bit_f_cnt` in the add device wizard. I see that in the v2 console it is not part of the add device form, only settings. Should we keep it in general settings only in v3 console as well?

kschiffer, 2020-09-30:
This should be part of the network layer settings in the wizard and the general settings form.

PR #3302 (2020-10-05): Set frame counter width in the Console

#### Summary

Closes #2438 

#### Changes

- Add `supports_32_bit_f_ct` to device wizard network settings
- Add `supports_32_bit_f_ct` to device general settings: network section
- Fix bug with `encode` function in `Field` component ( https://github.com/TheThingsNetwork/lorawan-stack/pull/3302#discussion_r498175002 )

#### Testing

- Setting new devices frame counter width
- Checking existing devices frame counter width

##### Regressions

- It is a new feature, the only regression is if the general settings or network part of device wizard don't handle  [... cut]
closed 2020-10-05 by pgalic96