TheThingsNetwork/lorawan-stack #2440: Setting indoor/outdoor location flag for gateways

kschiffer, 2020-04-29:
#### Summary
The console should allow setting whether a gateway antenna is located inside or outside.

#### Why do we need this?
Closing v2/v3 feature gap.

#### What is already there? What do you see now?
Location settings subview for gateways

#### What is missing? What do you want to see?
- A way to store this information in the gateway record (should be per antenna)
- A checkbox to set the location as indoor or outdoor

#### How do you propose to implement this?
We first need to update the gateway registry, then it's a matter of adding a checkbox in the location tab of the gateway.

#### Can you do this yourself and submit a Pull Request?
@pgalic96, though we should agree on whether this should indeed be added and how where to store the info in the gateway registry, cc @htdvisser

htdvisser, 2020-04-29:
Since this is just metadata (it's not actually used by The Things Stack) it can just be an attribute on the GatewayAntenna object. 

I think this is mostly just copy-paste of #2134. Maybe we can define a list of well-known attribute keys and common values for each of these, similar to what we have in the v2 console with the gateway brand/model fields.

htdvisser, 2020-08-24:
Not important enough for next up. If someone from the community wants to pick this up, let us know!