TheThingsNetwork/lorawan-stack #2441: Setting and displaying gateway manufacturer info in the Console

kschiffer, 2020-04-29:
#### Summary
The console should allow to store and display relevant manufacturer info (brand, model, antenna(s)) of a gateway.

#### Why do we need this?
- Provide relevant and helpful information of gateways
- Closing v2/v3 feature gap

#### What is already there? What do you see now?
Setting and reviewing info such as gateway description, name and id.

#### What is missing? What do you want to see?
- Setting manufacturer info
- Displaying that info in the gateway list and gateway overview

#### How do you propose to implement this?
We already have message types for gateway brand, model, version and radio. From first glance, I cannot see how these are meant to be attached to the gateway record (cc @htdvisser).
Also, I believe we plan to partially gather this info automatically via gateway repositories (?). Some clarification here would be helpful.

#### Can you do this yourself and submit a Pull Request?
Yes. Though anyone feel free to pick up.

htdvisser, 2020-04-29:
This was indeed designed around the gateway repository. The `brand_id`, `model_id`, `hardware_version` and `firmware_version` fields correspond to entries in that repository. As far as I know, no work has been done on the gateway repository since we designed the API.

johanstokking, 2020-08-24:
Blocked by https://github.com/TheThingsIndustries/lorawan-stack/issues/2018