TheThingsNetwork/lorawan-stack #2442: Decode PHYPayload before forwarding uplink message

johanstokking, 2020-04-29:
#### Summary

References #1962 

Decode PHYPayload before forwarding uplink message

#### Why do we need this?

For event subscribers to see LoRaWAN identifiers, i.e. EUIs in join-requests and DevAddr in uplink messages.

#### What is already there? What do you see now?

Basic Station has the `(*UplinkMessage).Payload` set, because it reconstructs the `RawPayload`. This is obviously not the case on other frontends (UDP, MQTT), so `Payload` is unset.

#### What is missing? What do you want to see?

Make sure that `Payload` is set.

#### How do you propose to implement this?

- Check if `Payload` is set. If it's set, fine, done
- _Try_ to decode the LoRaWAN message. That may fail. If it does, log a debug level message and still forward the message. The reason for this is that GS is not responsible for LoRaWAN handling; the `Payload` is informative only; NS will always try to decode the payload at all times

#### Can you do this yourself and submit a Pull Request?

Can review

KrishnaIyer, 2020-04-29:
I think it's better to do this in the frontends to keep it consistent with Basic Station. So that would 
decoding before calling `conn.HandleUp` in each of the frontends.

johanstokking, 2020-04-29:
It would be repeated there though, with potentially converging behavior on when that decoding fails, i.e. different error handling, log message, log level, etc. If we see it as part of the event payload, it's a GS-level thing. It's just that Basic Station magically knows the payload already (because that is the protocol).

neoaggelos, 2020-05-03:
So, it seems that the right place for this would be here?

https://github.com/TheThingsNetwork/lorawan-stack/blob/08194c375b439396612436e09529d112948bf024/pkg/gatewayserver/gatewayserver.go#L608

However, I can see that GS already gets retrieves some of that info here:

https://github.com/TheThingsNetwork/lorawan-stack/blob/08194c375b439396612436e09529d112948bf024/pkg/gatewayserver/gatewayserver.go#L551

So it seems to me that GS already has the devaddr, EUIs, etc. Is this about including these in the events instead?

johanstokking, 2020-05-05:
> So, it seems that the right place for this would be here?

Yes that is the place

> However, I can see that GS already gets retrieves some of that info here:

Yes but that is later

> So it seems to me that GS already has the devaddr, EUIs, etc. Is this about including these in the events instead?

Yes but the events do not allow for device identifiers, only for LoRaWAN payload where the identifiers are in. So I'd propose;

1. Decode the LoRaWAN payload in the section at L608
2. Replace decoding the identifiers by looking at the already decoded payload to get the identifiers

PR #2479 (2020-05-07): Decode LoRaWAN Payload in Gateway Server before pushing message

#### Summary

Closes #2442

#### Changes

#### Notes for Reviewers

I get a feeling that we do not get a lot out of having both `GetUplinkMessageIdentifiersFromMessage` and `GetUplinkMessageIdentifiers`.

Targetting `develop`, just because we are changing the current behaviour (we are only adding things though)

#### Checklist

- [X] Scope: The referenced issue is addressed, there are no unrelated changes.
- [X] Compatibility: The changes are backwards compatible with existing API, storage, configuration and CLI, according to the [... cut]
closed 2020-05-07 by johanstokking