TheThingsNetwork/lorawan-stack #2444: Allow admins to see all gateways and applications in the tenant

laurensslats, 2020-04-29:
#### Summary
Admins need to be able to know everything that's happening on their network server, as of now it is not possible to get a clear overview (in the console)
...

#### Why do we need this?
Admins need to know where the network server is used for and what devices/gateways are connected. 

#### What is already there? What do you see now?
Admins already have the rights, but have no (easy) way of finding out the names of applications/gateways. 
...

#### What is missing? What do you want to see?
Proposing something like this.

By default, the gateways and applications are not shown, by clicking the button, all gateways are shown, also the ones where the admin user is not a collaborator of.
[image]
...

laurensslats, 2020-05-26:
@kschiffer can you explain what you mentioned during the V3 meeting (the usage of "Tabs")

johanstokking, 2020-05-26:
I'd suggest the following:

- API: Add toggle `include_not_collaborator` (bool) to list requests (like https://github.com/TheThingsNetwork/lorawan-stack/issues/1903)
- IS: Update entity stores to query entities based on the toggle

For display, it we can do a tab, or a checkbox. I'm fine with either.

kschiffer, 2020-05-27:
> @kschiffer can you explain what you mentioned during the V3 meeting (the usage of "Tabs")

I was talking about tabbed tables like this:
[image]
(disregard discrepancy of the table columns with the current implementation)

We already have logic to implement this and it's a clearer separation IMO.

htdvisser, 2020-06-02:
Adding to **What is already there**:

- We have the `/api/v3/applications` API that lists applications the current user is a direct or indirect collaborator on.
- We have the `/api/v3/users/{user_id}/applications` and `/organizations/{organization_id}/applications` API to list applications the given user or organization (directly) collaborates on
- We have the `/api/v3/search/applications` API to search for applications by ID/Name/Description substring.
	- To admins this API already already returns all applications when no filters are configured.
- We have similar API for gateways.

---

Adding

htdvisser, 2020-06-02:
Sorry about that.

**How do you propose to implement this:**

I propose to use the search API, since this already supports the use-case of admins searching for all registered applications (or gateways).

closed 2020-06-23 by bafonins