TheThingsNetwork/lorawan-stack #2446: Reset MAC state when device uplink frame counter decreased

rvolosatovs, 2020-04-29:
#### Summary

Reset MAC state when device uplink frame counter decreased

#### Why do we need this?

To support FCntUp resets

#### What is already there? What do you see now?

Just set FCntUp

#### What is missing? What do you want to see?

MAC state should be reset when FCntUp of a device is set to lower value via registry

#### How do you propose to implement this?

Just like NS already does when matching

#### Can you do this yourself and submit a Pull Request?

yes

htdvisser, 2020-10-27:
This prevents some users from migrating from v2 to v3, so setting medium priority.

rvolosatovs, 2020-10-27:
@johanstokking coming back to the discussion we had during sync, this issue addresses specifically decreasing FCnt via device registry and resetting MAC state in response to that, such that a button in the console would then set the FCnt of the device to 0 or other value

rvolosatovs, 2020-11-03:
But taking another look at this, it makes much more sense to introduce a separate RPC for this. PR's coming soon!

closed 2020-12-28 by rvolosatovs