TheThingsNetwork/lorawan-stack #2448: Convert common/well-known errors to defined errors

htdvisser, 2020-04-29:
#### Summary

We should convert some common and well-known errors to defined errors.

#### Why do we need this?

To give more insight into what is going on when an error is returned.

#### What is already there? What do you see now?

If during OAuth token exchange a DNS error, timeout or other network error occurs, an internal server error is returned without details.

#### What is missing? What do you want to see?

Some details about the network error. Maybe DNS couldn't be resolved? Maybe the TCP connection timed out? Maybe the TLS expired? ...

#### How do you propose to implement this?

We could make `errors.From` recognize some common errors and return rich errors with attributes (built from definitions).

#### Can you do this yourself and submit a Pull Request?

Yes, but if anyone else would like to do this, just comment and re-assign.

neoaggelos, 2020-05-23:
Picking this one up

neoaggelos, 2020-06-02:
Closed with #2582

PR #2582 (2020-05-26): all: Convert some standard errors into well known errors

#### Summary

Closes #2448

#### Changes

- Well defined errors for context canceled and deadline exceeded
- `errors.From` can parse most common net.Errors

#### Testing

unit tests

##### Regressions

...

#### Notes for Reviewers

#### Checklist

- [X] Scope: The referenced issue is addressed, there are no unrelated changes.
- [ ] Compatibility: The changes are backwards compatible with existing API, storage, configuration and CLI, according to the compatibility commitments in `README.md`. The target branch is set to `master` if the c [... cut]
closed 2020-06-02 by neoaggelos