TheThingsNetwork/lorawan-stack #2454: Document pub/sub integrations

johanstokking, 2020-04-29:
#### Summary

Document application pub/sub integrations

#### Why do we need this ?

To complete the application data integrations

#### What is already there? What do you see now?

MQTT Server and HTTP Webhooks

#### What is missing? What do you want to see?

PubSub integrations

#### How do you propose to document this?

Add section like MQTT Server and Webhooks

It's good to mention also that we have two sorts of MQTT integrations; we have a builtin server (MQTT Server), and we have a client (via PubSub)

#### Can you do this yourself and submit a Pull Request?

Can review

benolayinka, 2020-05-05:
Working on this now. Is there any way to test the PubSub integration without your own NATS server?
In the Webhook guide we suggest using Postbin, is there something like that for NATS?

benolayinka, 2020-05-05:
> Working on this now. Is there any way to test the PubSub integration without your own NATS server?
> In the Webhook guide we suggest using Postbin, is there something like that for NATS?

@adriansmares

johanstokking, 2020-05-05:
I'd go with a local MQTT server (Mosquitto) or NATS. It's pretty easy to setup.

PR #2505 (2020-05-19): Add PubSub integration guide to docs

#### Summary

Closes #2454 

#### Changes

- Add PubSub integration guide to docs

#### Notes for Reviewers

I couldn't get mbed-simulator to work, so next week I will run thru this guide using a real gateway connected to my dev stack. Pending self-review.

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API, storage, configuration and CLI, according to the compatibility commitments in `README.md`. The target branch is set to `master` if  [... cut]
closed 2020-05-27 by benolayinka