TheThingsNetwork/lorawan-stack #2463: Improve payload formatters UI in the Console

bafonins, 2020-04-30:
#### Summary

I think it could be beneficial to change UI of payload formatters in the Console.

#### Why do we need this?

Better user experience. Currently, it is quite difficult to determine which formatter is used for a particular end device when both app/device formatters are set. Also, payload formatters options are not documented anywhere which is confusing (`None` vs `Repository`?). Additionally, refactoring the code for payload formatters (in the Console) would be nice as well.

#### What is already there? What do you see now?

[image]

#### What is missing? What do you want to see?

1. Inform the user when on the end device payload formatters page if there is app formatter set and which
2. Inform the user which payload formatter is used for a particular device
3. Overall improved UI
4. Separate UI's for app/device payload formatters pages

#### Environment

v3.7.2

#### Can you do this yourself and submit a Pull Request?

yes

@laurensslats  @benolayinka  - documentation, suggestions, feedback
@kschiffer  @pierrephz - design, suggestions

cc @johanstokking

kschiffer, 2020-05-01:
Implementation should be similar to
[image]

kschiffer, 2020-05-06:
I've put this into the current milestone since the current implementation being rather `bug`-ish and should hence be addressed soon.

As said in the closed PR, I am also fine with using the [initial solution](https://github.com/TheThingsNetwork/lorawan-stack/pull/2462) as long as we add some disclaimer/warning about the issue of deleting the payload formatters altogether.

@bafonins do you have time to work on this in May or should we reassign? (cc @pgalic96 )

pgalic96, 2020-06-14:
The issue with resetting device payload formatters is closed via https://github.com/TheThingsNetwork/lorawan-stack/pull/2695 , however the UI still needs to be improved via refactoring of the whole form.

benolayinka, 2020-06-16:
> Implementation should be similar to
> [image]
> 
> We need a global switch because setting the payload formatter to none will overwrite the formatters for both directions. This implication needs to be reflected in the UI. In the solution depicted in my screendesign, we can also get rid of the info notification since the situation should be clear enough with the radio button options.
> 
> Additionally, there should be a warning displayed when the user sets the `Usage type` from end device to application default, since the end device level formatter settings will not be stored.

Why isn't the end device specific formatter stored? It could be nice to switch between them temporarily.

kschiffer, 2020-06-19:
There is no persistence of that data in the backend. When the "None" option is chosen, the registry value is overwritten with an empty object. In order to still save the old value, we'd need to store it somewhere else, likely the `localStorage` or the global redux state of the application.

We can (and should) look into doing such things, but need to find a good mechanism for storing such arbitrary data. cc @bafonins

PR #5136 (2022-02-22): Improve PF UI

#### Summary

- Closes [#691](https://github.com/TheThingsIndustries/lorawan-stack-support/issues/691)
- Closes #2463 

#### Changes

- Add paste buttons
[image]
[image]

- Change notification position and add information about the app PF
[image]
[image]

#### Testing

e2e tests and manually

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes
closed 2022-02-22 by ryaplots