TheThingsNetwork/lorawan-stack #2468: Large number of MAC commands

ktsolakas, 2020-05-01:
#### Summary
On our TTI instance we have a customer that has added about 300 class C ABP devices. The customer is provisioning each controller one at a time and at this provisioning stage only one controller is active at a time. During the provisioning the customer is experiencing errors and inconsistencies while sending downlinks. When looking at the TTI logs i see a huge number of mac commands. I will attach a small section of the logs (~20sec) in which TTI generated 72000 lines which contain MAC commands (
[TTI_debug_logs.zip](https://github.com/TheThingsNetwork/lorawan-stack/files/4562346/TTI_debug_logs.zip)
TTI_debug_logs.zip). When looking at the Gateway logs i see a large amount of downliks being sent with a large percentage of those downlinks being marked as rejected due to collisions. Attaching the gateway logs as well (lora-pkt-fwd-1.zip)
[lora-pkt-fwd-1.zip](https://github.com/TheThingsNetwork/lorawan-stack/files/4562350/lora-pkt-fwd-1.zip
). I am not sure if the 2 are related to each other but what i am looking for is the reason we are seeing so many MAC commands, so many downlinks on the Gateway and  a way to avoid collisions on the Gateway.

#### Steps to Reproduce

1. Add class C ABP device
2. Turn on device 
3. Turn off device
4. NS generates MAC commands constantly even when device is offline for a large amount of time. Gateway is sending alot of downlinks many of which are indicated as collisions.

If you use the command "ttn-lw-stack config", you can redact sensitive config.
[TTI_config.txt](https://github.com/TheThingsNetwork/lorawan-stack/files/4562353/TTI_config.txt)

#### What do you want to see instead?
Want to see debug logs that are not polluted by mostly MAC commands. And want to see no collisions on the Gateway.

#### Environment
ttn-lw-stack version = 3.7.0

#### How do you propose to implement this?
Do not know

#### Can you do this yourself and submit a Pull Request?
No

rvolosatovs, 2020-05-06:
Thank you for reporting, looking at the logs, I see that the paths for class B/C downlink are missing, so Network Server keep retrying, that will be fixed by https://github.com/TheThingsNetwork/lorawan-stack/pull/2418

rvolosatovs, 2020-05-12:
Closed by https://github.com/TheThingsNetwork/lorawan-stack/pull/2418

closed 2020-05-12 by rvolosatovs

johanstokking, 2020-06-05:
@rvolosatovs we got a report that since 3.8.2 this issue persisted for 6 out of 300 devices. It was only resolved after recreating the devices. Any ideas what that might have caused?

rvolosatovs, 2020-06-08:
Probably corrupted MAC states - are the devices ABP?

ama9910, 2020-06-09:
Yes, ABP.

johanstokking, 2020-06-09:
@ama9910 next time please get us the output of;

```bash
$ ttn-lw-cli end-devices get <app-id> <dev-id> --all
```

And redact for sensitive data (keys and identifiers)

rvolosatovs, 2020-06-09:
> Yes, ABP.

That's the reason then, NS needs to recompute the MAC state and that only happens on FCnt reset and/or recreation for ABP in 1.0.x.
Note that FCnt(and, hence, MAC state) resets will be supported via console in the future: https://github.com/TheThingsNetwork/lorawan-stack/issues/2446