TheThingsNetwork/lorawan-stack #2473: LoRaCloud DAS package should send join event to the DAS

Oliv4945, 2020-05-04:
#### Summary
LoRa Cloud API evolved in February so application package should call DAS on join event. It does not matter if it is on joinReq or joinAccept

#### Why do we need this?
It allows the DAS to reset context associated to a device

#### What is missing? What do you want to see?
As [per documentation](https://www.loracloud.com/documentation/device_management?url=v1.html#deliver-uplink-message), the message send to the DAS at `/api/v1/uplink/send` should be
```
{
  DEVEUI:  {
     "msgtype":   "joining"       // Required.
     "timestamp": FLOAT,         // Optional. RX timestamp in seconds UTC
     "dr":        UINT1,         // Optional. Data rate
     "freq":      UINT4,         // Optional. Frequency
  }
}
```

#### Can you do this yourself and submit a Pull Request?

No

adriansmares, 2020-05-06:
@Oliv4945 is this required for the initial release ? (the CLI-only one).

Oliv4945, 2020-05-06:
Yes please, so the DAS is aware of new sessions

PR #2539 (2020-05-15): Support application wide application packages associations

#### Summary

Closes #2412
Closes #2473
Closes #2474

#### Changes

- Add application packages default associations. These are the application-wide equivalent of normal application packages associations. They allow users to setup an application package for the whole application instead of an per-device basis. 
- Application packages may now handle _unrelated_ traffic (traffic on a different FPort than the one they've been associated with). This allows application packages to schedule downlinks even when their FPort hasn't been used.
- [... cut]
closed 2020-05-15 by adriansmares