TheThingsNetwork/lorawan-stack #2474: LoRaCloud DAS package uplinks management change

Oliv4945, 2020-05-04:
#### Summary
As [per documentation](https://www.loracloud.com/documentation/device_management?url=v1.html#post--api-v1-uplink-send) DAS should be called for all uplinks, so DAS gets all downlinks opportunity. 
For all non device management traffic, payload should be removed to ensure data privacy

#### What is missing? What do you want to see?

For all traffic not on DAS port registered by the user when creating the application package association, the uplink should be forwarded to the DAS:

```
{
  DEVEUI:  {
    "msgtype":   "updf"    // Optional. 'updf' is implied if not present
    "fcnt":      UINT4,    // Required. Frame counter
    "port":      UINT1,    // Required. Port
    "payload":   HEX,      // Required. HEX string with valid modem payload if port = device management port
    "dr":        UINT1,    // Optional. Data rate.
    "freq":      UINT4,    // Optional. Frequency
    "timestamp": FLOAT     // Optional. RX timestamp in seconds, UTC
    "dn_mtu":    UINT4,    // Optional. Downlink MTU
  },
}
```

#### Can you do this yourself and submit a Pull Request?

Maybe, although I might require help to implement tests

adriansmares, 2020-05-06:
@Oliv4945 is this required for the initial release ? (the CLI-only one).

Oliv4945, 2020-05-06:
Yes it is required, otherwise it will not work for people using non default port

PR #2539 (2020-05-15): Support application wide application packages associations

#### Summary

Closes #2412
Closes #2473
Closes #2474

#### Changes

- Add application packages default associations. These are the application-wide equivalent of normal application packages associations. They allow users to setup an application package for the whole application instead of an per-device basis. 
- Application packages may now handle _unrelated_ traffic (traffic on a different FPort than the one they've been associated with). This allows application packages to schedule downlinks even when their FPort hasn't been used.
- [... cut]
closed 2020-05-15 by adriansmares