TheThingsNetwork/lorawan-stack #2475: Refactor Network Server downlink tests

rvolosatovs, 2020-05-04:
#### Summary

Refactor Network Server downlink tests

#### Why do we need this?

In their current form Network Server downlink tests are hard to maintain and incomplete

#### What is already there? What do you see now?

Very verbose, ad-hoc tests

#### What is missing? What do you want to see?

Now that Network Server structure/API are quite stable and requirements are clear, the tests can be improved

#### How do you propose to implement this?

Same as was just done for uplink tests - introduce assertions, make reasonable assumptions for simplicity.

#### Can you do this yourself and submit a Pull Request?

yes

nejraselimovic, 2021-06-07:
@htdvisser should this be a part of #3142 ?

htdvisser, 2021-06-08:
Maybe @adriansmares can better answer that

PR #3052 (2020-09-10): Refactor Network Server testing

#### Summary

Refactor NS testing. Blocking #2837 
Refs https://github.com/TheThingsNetwork/lorawan-stack/issues/2732
Closes #2475 

#### Changes

- Remove internal mocks where possible (more to come in #2837)
- Use Redis registry internally for testing (when we have other backends implemented, we can make this configurable, e.g. via env variables)
- Remove highly-specific uplink tests - the functionality they tested is now sufficiently covered by flow tests. Adapting those tests would take enormous amount of additional time and I have spent enough of time on t [... cut]