TheThingsNetwork/lorawan-stack #2497: After upgrading to v3.8.0 /healthz/ready returns 404

kurtmc, 2020-05-08:
#### Summary

We have upgraded from v3.7.2 to v3.8.0 and now /healthz/ready endpoint is returning 404.

#### Steps to Reproduce

1. Run lorawan-stack v3.8.0 with `TTN_LW_HTTP_HEALTH_ENABLE=true`
2. curl http://localhost:1885/healthz/ready

#### What do you see now?

```
404 page not found
```

Sorry that this report is light on details but I think it should be fairly simple to reproduce, if not I will create a docker-compose.yml that show the issue.

johanstokking, 2020-05-08:
Thanks for reporting, we'll look at it!

htdvisser, 2020-05-25:
This should be fixed now that https://github.com/TheThingsNetwork/lorawan-stack/pull/2519 is merged. These fixes will be part of the next release.

closed 2020-05-25 by htdvisser