TheThingsNetwork/lorawan-stack #2500: Issues with PubSub

laurensslats, 2020-05-08:
#### Summary

Issues with PubSub (in AWS AIM)
- “fail” messages are received non-stop, which don’t seems to be related to incoming uplinks.
 - I’ve attached a screenshot of my console to show what I see
 - The JSON error message i’ve inserted at the bottom of this email.
 - The uplinks are all getting into AWS IoT, so PubSub is working somehow.
- Enabling/Disabling Messages types doesn’t seem to do anything
- Only Uplink and Join Accept messages make it to the PubSub destination.
- Base Topic seems to like it’s not implemented yet.
- Gateway Status updates are not a message type option yet.

...

#### Steps to Reproduce
1. ...
2. ...
3. ...

#### What do you see now?
[image]

**PubSub Fail Message Example:**
```
{
  "@type": "type.googleapis.com/ttn.lorawan.v3.ErrorDetails",
  "namespace": "pkg/applicationserver/io/pubsub",
  "name": "integration_failed",
  "message_format": "integration {pub_sub_id} failed",
  "attributes": {
    "pub_sub_id": "aws-iot-integration"
  },
  "cause": {
    "namespace": "pkg/applicationserver/io/pubsub/provider/mqtt",
    "name": "connect_failed",
    "message_format": "connection to MQTT server failed",
    "code": 4
  },
  "code": 10
}
```

**Gateway Drop Uplink Message Example:**
```
{
  "@type": "type.googleapis.com/ttn.lorawan.v3.ErrorDetails",
  "namespace": "pkg/gatewayserver",
  "name": "host_handle",
  "message_format": "host `{host}` failed to handle message",
  "attributes": {
    "host": "cluster"
  },
  "cause": {
    "namespace": "pkg/networkserver",
    "name": "device_not_found",
    "message_format": "device not found",
    "correlation_id": "a064d4e306d049ddb8b7af5099b6ebdb",
    "code": 5
  },
  "code": 5
}
```

...

#### What do you want to see instead?

...

#### Environment
v3.7.0
...

#### How do you propose to implement this?

...

#### Can you do this yourself and submit a Pull Request?

...

johanstokking, 2020-05-08:
@laurensslats please fill steps to reproduce

johnwiles, 2020-05-26:
Steps To Reproduce:
0. Create New Application (with no devices)
1. Add Integration: PubSub
2. Select Format: JSON
3. Select any Base Topic
4. Select Provider: MQTT
5. Secure: Enabled
6. Input Security Files, URL and Client-ID
7. Use Credentials: Unchecked
8. Select QOS 0 for both
9. Save Changes

View "Data" window of application, see "Fail" messages.
One is added every 18 seconds, continually:
[image]

johanstokking, 2020-05-26:
@johnwiles can you verify you can connect with for example `mosquitto_sub` to the MQTT server with the same credentials? Is it an external server, or is it running locally?

johnwiles, 2020-05-26:
@johanstokking I haven't tested with a different MQTT client. I will give that a go and report back.
It's an external server: AWS IoT Core.

But It is seemingly connecting to AWS IoT Core all fine; on a production application (not the application in screenshot above) the payloads are arriving at AWS IoT Core, but, the error messages keep coming up in the data stream anyway.
Then with a blank application (no devices), we still get these error messages every 18 seconds non stop. Even when I change the publish URL, the security, the payload format, anything...no matter what I'm seeing the exact same error messages every 18 seconds.

More verbose error messages would be very useful.

adriansmares, 2020-10-01:
> @johanstokking I haven't tested with a different MQTT client. I will give that a go and report back.
> It's an external server: AWS IoT Core.
>
> But It is seemingly connecting to AWS IoT Core all fine; [...]

Is this issue still present in the last release ?

johanstokking, 2020-10-02:
Please reopen if still applicable.

closed 2020-10-02 by johanstokking