TheThingsNetwork/lorawan-stack #2504: MQTT Server not reachable/not running

laubed, 2020-05-08:
#### Summary
I used the Getting started guide with docker. Doing so results in a usable stack except for MQTT server. It seems like there is nothing listening on that ports (1881,8881,1883,8883). Connections to these ports result in timeouts.

#### Steps to Reproduce
1. Install the stack using the Getting started guide with docker.
2. Try ton connect to the MQTT server.
3. (Optional) create an api key for connecting to the mqtt server. Same result here.

#### Environment
Release 3.8 on Centos 8

johanstokking, 2020-05-08:
What is your Gateway Server configuration exactly?

laubed, 2020-05-08:
```
gs:
  mqtt:
    public-address: 'mydomain:1882'
    public-tls-address: 'mydomain:8882'
  mqtt-v2:
    public-address: 'mydomain:1881'
    public-tls-address: 'mydomain:8881'
```

where mydomain ist my external domain name.

KrishnaIyer, 2020-05-09:
If you haven't explicitly set the listeners for the MQTT servers then the [default values](https://thethingsstack.io/latest/reference/networking/) will be used.

Can you dump your config here;
```
$ docker-compose run --rm stack config
```
And make sure to retract sensitive information (ex: passwords, domain config etc.)

laubed, 2020-05-10:
I think i'll stripped als sensitive information.

```
                                           --as.device-kek-label=""
                                        --as.interop.blob.bucket=""
                                          --as.interop.blob.path=""
                                      --as.interop.config-source=""
                                          --as.interop.directory=""
                                                 --as.interop.id=""
                                                --as.interop.url=""
                                                  --as.link-mode="all"
                                                --as.mqtt.listen=":1883"
                                            --as.mqtt.listen-tls=":8883"
                                        --as.mqtt.public-address="localhost:1883"
                                    --as.mqtt.public-tls-address="localhost:8883"
                           --as.webhooks.downlink.public-address="http://localhost:1885/api/v3"
                       --as.webhooks.downlink.public-tls-address=""
                                        --as.webhooks.queue-size="16"
                                            --as.webhooks.target="direct"
                               --as.webhooks.templates.directory=""
                           --as.webhooks.templates.logo-base-url=""
                                     --as.webhooks.templates.url=""
                                           --as.webhooks.timeout="5s"
                                           --as.webhooks.workers="16"
                                        --blob.aws.access-key-id=""
                                             --blob.aws.endpoint=""
                                               --blob.aws.region=""
                                    --blob.aws.secret-access-key=""
                                        --blob.aws.session-token=""
                                          --blob.gcp.credentials=""
                                     --blob.gcp.credentials-file=""
                                          --blob.local.directory="/srv/ttn-lorawan/public/blob"
                                                 --blob.provider="local"
                                           --cache.redis.address=""
                                          --cache.redis.database="0"
                                --cache.redis.failover.addresses=""
                                   --cache.redis.failover.enable="false"
                              --cache.redis.failover.master-name=""
                                         --cache.redis.namespace=""
                                          --cache.redis.password=""
                                         --cache.redis.pool-size="0"
                                                 --cache.service=""
                                               --cluster.address=""
                                    --cluster.application-server=""
                                         --cluster.crypto-server=""
                                        --cluster.gateway-server=""
                                       --cluster.identity-server=""
                                                  --cluster.join=""
                                           --cluster.join-server=""
                                                  --cluster.keys=""
                                                  --cluster.name=""
                                        --cluster.network-server=""
                                                   --cluster.tls="false"
                                                        --config="/config/ttn-lw-stack.yml"
                                                 --console.mount=""
                                   --console.oauth.authorize-url="https://mydomain/oauth/authorize"
                                       --console.oauth.client-id="console"
                                   --console.oauth.client-secret="oauthsecret"
                                      --console.oauth.logout-url="http://localhost:1885/oauth/logout"
                                       --console.oauth.token-url="https://mydomain/oauth/token"
                                        --console.ui.as.base-url="https://mydomain/api/v3"
                                         --console.ui.as.enabled="true"
                                    --console.ui.assets-base-url="/assets"
                                  --console.ui.branding-base-url=""
                                      --console.ui.canonical-url="https://mydomain/console"
                                           --console.ui.css-file="console.css"
                                       --console.ui.descriptions=""
                             --console.ui.documentation-base-url="https://thethingsstack.io/3.8.0"
                                      --console.ui.edtc.base-url="https://mydomain/api/v3"
                                       --console.ui.edtc.enabled="true"
[... 190 lines cut ...]
                                          --pba.tls.key-vault.id=""
                                                --pba.tls.source=""
                                                 --redis.address="redis:6379"
                                                --redis.database="0"
                                      --redis.failover.addresses=""
                                         --redis.failover.enable="false"
[... cut ...]
                                                --tls.acme.hosts="mydomain"
                                               --tls.certificate="cert.pem"
                                      --tls.insecure-skip-verify="false"
                                                       --tls.key="key.pem"
                                              --tls.key-vault.id=""
                                                   --tls.root-ca=""
                                                    --tls.source="acme"
```

KrishnaIyer, 2020-05-10:
Yeah these settings are correctly using the defaults:
```
                                             --gs.mqtt-v2.listen=":1881"
                                         --gs.mqtt-v2.listen-tls=":8881"
                                     --gs.mqtt-v2.public-address="mydomain:1881"
                                 --gs.mqtt-v2.public-tls-address="mydomain:8881"
                                                --gs.mqtt.listen=":1882"
                                            --gs.mqtt.listen-tls=":8882"
                                        --gs.mqtt.public-address="mydomain:1882"
                                    --gs.mqtt.public-tls-address="mydomain:8882"
```

And what ports are you binding to the container? (Basically the `ports` section of `docker-compose.yml`)

laubed, 2020-05-10:
I did not change any ports (except for the 1885 and 8885 ones)

```
    ports:
      # If deploying on a public server:
      - "80:1885"
      - "443:8885"
      - "1881:1881"
      - "8881:8881"
      - "1882:1882"
      - "8882:8882"
      - "1883:1883"
      - "8883:8883"
      - "1884:1884"
      - "8884:8884"
      #- "1885:1885"
      #- "8885:8885"
      - "1887:1887"
      - "8887:8887"
      - "1700:1700/udp"
```

laubed, 2020-05-10:
It seems like i can connect to 8883 and 1883 now with application keys fine. But not on any other port.
For what are the other mqtt ports?

KrishnaIyer, 2020-05-11:
The other mqtt ports are for MQTT gateways to connect. The docker configuration seems correct.
Can you see if the ports are being bound properly by the container, basically the output of 
```
$ docker ps -a | grep stack
```

johanstokking, 2020-05-29:
This seems to be working as expected, closing this one. Please reopen if the issue persists, providing details.

closed 2020-05-29 by johanstokking