TheThingsNetwork/lorawan-stack #2521: Console - Forbidden token exchange refused

pablorq, 2020-05-12:
#### Summary

After install, the first login fails with "Forbidden token exchange refused" error.

#### Steps to Reproduce

1. Install the stack following the [installation guide](https://thethingsstack.io/v3.8.1/getting-started/installation/)
2. Run the container:
```
$ sudo docker-compose up
```
3. Open the console
```
http://localhost:1885
```
4. Login with admin
5. "Forbidden token exchange refused" error

#### What do you see now?

A web page wiht "Forbidden token exchange refused" error.
```
http://localhost:1885/console/oauth/callback?code=MF2XI.I7FYJJHQNR5R6P3YHXVFZ6XCF2B2HRVJGHD2DIA.YYHMRDIFHZAGGNGE3TRLNTAJTFS6X33SPSNVITZFRUS3HBKLM76A&state=i_srwhL8tFJkb-j2
```

#### What do you want to see instead?

The console web page.

#### Environment

Ubuntu 19.10
```
$ sudo docker --version
Docker version 19.03.8, build afacb8b7f0
$ sudo docker-compose --version
docker-compose version 1.21.0, build unknown
```
URL: localhost
certificates: self-signed
<details>
<summary>docker-compose.yml</summary>

```text
version: '3.6'
services:

  # If using CockroachDB:
  cockroach:
    image: cockroachdb/cockroach
    command: start --http-port 26256 --insecure
    restart: unless-stopped
    volumes:
      - ${DEV_DATA_DIR:-.env/data}/cockroach:/cockroach/cockroach-data
    ports:
      - "127.0.0.1:26257:26257" # Cockroach
      - "127.0.0.1:26256:26256" # WebUI

  # If using PostgreSQL:
  # postgres:
  #   image: postgres
  #   restart: unless-stopped
  #   environment:
  #     - POSTGRES_PASSWORD=root
  #     - POSTGRES_USER=root
  #     - POSTGRES_DB=ttn_lorawan
  #   volumes:
  #     - ${DEV_DATA_DIR:-.env/data}/postgres:/var/lib/postgresql/data
  #   ports:
  #     - "127.0.0.1:5432:5432"

  redis:
    image: redis
    command: redis-server --appendonly yes
    restart: unless-stopped
    volumes:
      - ${DEV_DATA_DIR:-.env/data}/redis:/data
    ports:
      - "127.0.0.1:6379:6379"

  stack:
    image: thethingsnetwork/lorawan-stack
    entrypoint: ttn-lw-stack -c /config/ttn-lw-stack.yml
    command: start
    restart: unless-stopped
    depends_on:
      - redis
      # If using CockroachDB:
      - cockroach
      # If using PostgreSQL:
      # - postgres
    volumes:
      - ./blob:/srv/ttn-lorawan/public/blob
      - ./config/stack:/config:ro
      # If using Let's Encrypt:
      # - ./acme:/var/lib/acme
    environment:
      TTN_LW_BLOB_LOCAL_DIRECTORY: /srv/ttn-lorawan/public/blob
      TTN_LW_REDIS_ADDRESS: redis:6379
      # If using CockroachDB:
      TTN_LW_IS_DATABASE_URI: postgres://root@cockroach:26257/ttn_lorawan?sslmode=disable
      # # If using PostgreSQL:
      # TTN_LW_IS_DATABASE_URI: postgres://root:root@postgres:5432/ttn_lorawan?sslmode=disable

    ports:
      # If deploying on a public server:
      # - "80:1885"
      # - "443:8885"
      - "1881:1881"
      - "8881:8881"
      - "1882:1882"
      - "8882:8882"
      - "1883:1883"
      - "8883:8883"
      - "1884:1884"
      - "8884:8884"
      - "1885:1885"
      - "8885:8885"
      - "1887:1887"
      - "8887:8887"
      - "1700:1700/udp"

    # If using (self) signed certificates:
    secrets:
      - cert.pem
      - key.pem

# If using (self) signed certificates:
secrets:
  cert.pem:
    file: ./acme/cert.pem
  key.pem:
    file: ./acme/key.pem
```
</details>

<details>
<summary>ttn-lw-stack.yml</summary>

```text
version: '3.6'
services:

  # If using CockroachDB:
  cockroach:
    image: cockroachdb/cockroach
    command: start --http-port 26256 --insecure
    restart: unless-stopped
    volumes:
      - ${DEV_DATA_DIR:-.env/data}/cockroach:/cockroach/cockroach-data
    ports:
      - "127.0.0.1:26257:26257" # Cockroach
      - "127.0.0.1:26256:26256" # WebUI

  # If using PostgreSQL:
  # postgres:
  #   image: postgres
  #   restart: unless-stopped
  #   environment:
  #     - POSTGRES_PASSWORD=root
  #     - POSTGRES_USER=root
  #     - POSTGRES_DB=ttn_lorawan
  #   volumes:
  #     - ${DEV_DATA_DIR:-.env/data}/postgres:/var/lib/postgresql/data
  #   ports:
  #     - "127.0.0.1:5432:5432"

  redis:
    image: redis
    command: redis-server --appendonly yes
    restart: unless-stopped
    volumes:
      - ${DEV_DATA_DIR:-.env/data}/redis:/data
    ports:
      - "127.0.0.1:6379:6379"

  stack:
    image: thethingsnetwork/lorawan-stack
    entrypoint: ttn-lw-stack -c /config/ttn-lw-stack.yml
    command: start
    restart: unless-stopped
    depends_on:
      - redis
      # If using CockroachDB:
      - cockroach
      # If using PostgreSQL:
      # - postgres
    volumes:
      - ./blob:/srv/ttn-lorawan/public/blob
      - ./config/stack:/config:ro
      # If using Let's Encrypt:
      # - ./acme:/var/lib/acme
    environment:
      TTN_LW_BLOB_LOCAL_DIRECTORY: /srv/ttn-lorawan/public/blob
      TTN_LW_REDIS_ADDRESS: redis:6379
      # If using CockroachDB:
      TTN_LW_IS_DATABASE_URI: postgres://root@cockroach:26257/ttn_lorawan?sslmode=disable
      # # If using PostgreSQL:
      # TTN_LW_IS_DATABASE_URI: postgres://root:root@postgres:5432/ttn_lorawan?sslmode=disable

    ports:
      # If deploying on a public server:
      # - "80:1885"
      # - "443:8885"
      - "1881:1881"
      - "8881:8881"
      - "1882:1882"
      - "8882:8882"
      - "1883:1883"
      - "8883:8883"
      - "1884:1884"
      - "8884:8884"
      - "1885:1885"
      - "8885:8885"
      - "1887:1887"
      - "8887:8887"
      - "1700:1700/udp"

    # If using (self) signed certificates:
    secrets:
      - cert.pem
      - key.pem

# If using (self) signed certificates:
secrets:
  cert.pem:
    file: ./acme/cert.pem
  key.pem:
    file: ./acme/key.pem
```
</details>

#### How do you propose to implement this?

No idea.

#### Can you do this yourself and submit a Pull Request?

I don't think so. :(

pablorq, 2020-05-13:
### More info

#### Related issues

Issue #2353 
Not the same situation, because I'm not using CA certificates.

Issue #1818 
Some certificates problem but related to CLI.

#### Tests done

Accessing the container as root user:

```
$ sudo docker exec -it --user root tts_stack_1 ash
```

Running commands inside the container:

```
/ # whoami
root

/ # curl -v http://localhost:1885
*   Trying 127.0.0.1:1885...
* TCP_NODELAY set
* Connected to localhost (127.0.0.1) port 1885 (#0)
> GET / HTTP/1.1
> Host: localhost:1885
> User-Agent: curl/7.66.0
> Accept: */*
> 
* Mark bundle as not supporting multiuse
< HTTP/1.1 302 Found
< Location: /console
< Referrer-Policy: strict-origin-when-cross-origin
< Vary: Accept-Encoding
< X-Content-Type-Options: nosniff
< X-Frame-Options: SAMEORIGIN
< X-Request-Id: 01E84V9SB5KJF1MRXKKES8SBXK
< X-Xss-Protection: 1; mode=block
< Date: Tue, 12 May 2020 16:25:49 GMT
< Content-Length: 0
< 
* Connection #0 to host localhost left intact

/ # curl -v https://localhost:1885
*   Trying 127.0.0.1:1885...
* TCP_NODELAY set
* Connected to localhost (127.0.0.1) port 1885 (#0)
* ALPN, offering h2
* ALPN, offering http/1.1
* successfully set certificate verify locations:
*   CAfile: /etc/ssl/certs/ca-certificates.crt
  CApath: none
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
* error:1408F10B:SSL routines:ssl3_get_record:wrong version number
* Closing connection 0
curl: (35) error:1408F10B:SSL routines:ssl3_get_record:wrong version number
```

It seems that HTTP to localhost:1885 is ok, but HTTPS gives this error:

```
curl: (35) error:1408F10B:SSL routines:ssl3_get_record:wrong version number
```

Also tried to update the certificates

```
/ # update-ca-certificates 
WARNING: ca-certificates.crt does not contain exactly one certificate or CRL: skipping
```

This warning seems to be a known issue and a minor warning in Docker-Alpine as stated [here](https://github.com/gliderlabs/docker-alpine/issues/30). The warning is shown after updating the certificates.

pablorq, 2020-05-13:
By the way, I'm using the latest docker version from the [v3.8.1](https://github.com/TheThingsNetwork/lorawan-stack/releases/tag/v3.8.1) release

```
$ sudo docker images
REPOSITORY                       TAG                 IMAGE ID            CREATED             SIZE
cockroachdb/cockroach            latest              4a81e5f28a77        34 hours ago        238MB
thethingsnetwork/lorawan-stack   latest              11e7df28e933        42 hours ago        228MB
redis                            latest              f9b990972689        11 days ago         104MB
```

pablorq, 2020-05-13:
I've activated the ports 80 and 443 on the docker-compose.yml file:

```text
    ports:
      # If deploying on a public server:
      - "80:1885"
      - "443:8885"
      ...
```

And updated the stack:

```
$ sudo docker-compose up
```

<details>
<summary>docker-compose output</summary>

```
Starting tts_cockroach_1 ... done
Starting tts_redis_1     ... done
Recreating tts_stack_1   ... done
Attaching to tts_redis_1, tts_cockroach_1, tts_stack_1
redis_1      | 1:C 13 May 2020 08:14:27.038 # oO0OoO0OoO0Oo Redis is starting oO0OoO0OoO0Oo
redis_1      | 1:C 13 May 2020 08:14:27.039 # Redis version=6.0.1, bits=64, commit=00000000, modified=0, pid=1, just started
redis_1      | 1:C 13 May 2020 08:14:27.039 # Configuration loaded
redis_1      | 1:M 13 May 2020 08:14:27.041 * Running mode=standalone, port=6379.
redis_1      | 1:M 13 May 2020 08:14:27.042 # WARNING: The TCP backlog setting of 511 cannot be enforced because /proc/sys/net/core/somaxconn is set to the lower value of 128.
redis_1      | 1:M 13 May 2020 08:14:27.042 # Server initialized
redis_1      | 1:M 13 May 2020 08:14:27.042 # WARNING overcommit_memory is set to 0! Background save may fail under low memory condition. To fix this issue add 'vm.overcommit_memory = 1' to /etc/sysctl.conf and then reboot or run the command 'sysctl vm.overcommit_memory=1' for this to take effect.
redis_1      | 1:M 13 May 2020 08:14:27.042 # WARNING you have Transparent Huge Pages (THP) support enabled in your kernel. This will create latency and memory usage issues with Redis. To fix this issue run the command 'echo never > /sys/kernel/mm/transparent_hugepage/enabled' as root, and add it to your /etc/rc.local in order to retain the setting after a reboot. Redis must be restarted after THP is disabled.
redis_1      | 1:M 13 May 2020 08:14:27.042 * DB loaded from append only file: 0.000 seconds
redis_1      | 1:M 13 May 2020 08:14:27.042 * Ready to accept connections
cockroach_1  | *
cockroach_1  | * WARNING: RUNNING IN INSECURE MODE!
cockroach_1  | * 
cockroach_1  | * - Your cluster is open for any client that can access <all your IP addresses>.
cockroach_1  | * - Any user, even root, can log in without providing a password.
cockroach_1  | * - Any user, connecting as root, can read or write any data in your cluster.
cockroach_1  | * - There is no network encryption nor authentication, and thus no confidentiality.
cockroach_1  | * 
cockroach_1  | * Check out how to secure your cluster: https://www.cockroachlabs.com/docs/v20.1/secure-a-cluster.html
cockroach_1  | *
stack_1      |   INFO Setting up core component               
stack_1      |   INFO Setting up Identity Server              
cockroach_1  | *
cockroach_1  | * WARNING: running 'cockroach start' without --join is deprecated.
cockroach_1  | * Consider using 'cockroach start-single-node' or 'cockroach init' instead.
cockroach_1  | *
cockroach_1  | *
cockroach_1  | * WARNING: neither --listen-addr nor --advertise-addr was specified.
cockroach_1  | * The server will advertise "4ae0b4732828" to other nodes, is this routable?
cockroach_1  | * 
cockroach_1  | * Consider using:
cockroach_1  | * - for local-only servers:  --listen-addr=localhost
cockroach_1  | * - for multi-node clusters: --advertise-addr=<host/IP addr>
cockroach_1  | * 
cockroach_1  | *
cockroach_1  | CockroachDB node starting at 2020-05-13 08:14:31.119567741 +0000 UTC (took 4.0s)
cockroach_1  | build:               CCL v20.1.0 @ 2020/05/05 00:07:18 (go1.13.9)
cockroach_1  | webui:               http://4ae0b4732828:26256
cockroach_1  | sql:                 postgresql://root@4ae0b4732828:26257?sslmode=disable
cockroach_1  | RPC client flags:    /cockroach/cockroach <client cmd> --host=4ae0b4732828:26257 --insecure
cockroach_1  | logs:                /cockroach/cockroach-data/logs
cockroach_1  | temp dir:            /cockroach/cockroach-data/cockroach-temp548315742
cockroach_1  | external I/O path:   /cockroach/cockroach-data/extern
cockroach_1  | store[0]:            path=/cockroach/cockroach-data
cockroach_1  | storage engine:      rocksdb
cockroach_1  | status:              restarted pre-existing node
[... 122 lines cut ...]
    PSK identity hint: None
    SRP username: None
    TLS session ticket lifetime hint: 604800 (seconds)
    TLS session ticket:
    0000 - 1e b8 16 47 f1 b5 69 ca-fb 87 5d 3d 98 15 c2 98   ...G..i...]=....
    0010 - c5 db 6a 63 dd 77 5f 47-92 27 20 31 8a ab 32 23   ..jc.w_G.' 1..2#
    0020 - 5c 1a 2c 4b 9a 62 8a 90-eb 18 40 83 bd 71 80 23   \.,K.b....@..q.#
    0030 - 81 bb 9a 90 05 50 a7 0c-4d 76 d4 49 a5 59 61 89   .....P..Mv.I.Ya.
    0040 - 09 fe 63 f3 09 ae 99 3c-19 4e ed d1 69 e7 0f d8   ..c....<.N..i...
    0050 - 34 f9 c6 a3 e6 2a c3 67-4b 43 cc c3 4e 57 b8 32   4....*.gKC..NW.2
    0060 - e6 b3 78 34 db 15 d7 94-71 08 63 31 56 26 ad 96   ..x4....q.c1V&..
    0070 - 71                                                q

    Start Time: 1589361940
    Timeout   : 7200 (sec)
    Verify return code: 18 (self signed certificate)
    Extended master secret: no
    Max Early Data: 0
---
read R BLOCK
closed
```
</details>

Any idea?

johanstokking, 2020-05-14:
Thanks @pablorq for the extensive report.

> Issue #2353
> Not the same situation, because I'm not using CA certificates.

I do think it is related. The token exchange happens internally via TLS. Even if you're not using CA certificates, your self-signed certificates need to be trusted for the token exchange.

Please set your [TLS root CA](https://thethingsstack.io/v3.8.1/reference/configuration/the-things-stack/) to the server certificate. Yes, the CA is the self-signed certificate.

pablorq, 2020-05-14:
## New start - 200514

After looking at [this post](https://github.com/TheThingsNetwork/lorawan-stack/issues/2353#issuecomment-626498059) on #2353 issue I realized that my certificates have missing identifiers.

So I clean up all, create new certificates and started all over again. Also removed the certificate and cookies from the browser. In fact, at this point, I have some scripts to help me to do it for each new thing to test.

### Certificates
The attached certificates are the ones I've generated. Now I have a CA certificate and a cert.pem generated certificate with the correspondent key.pem file.

Certificates: [certs.zip](https://github.com/TheThingsNetwork/lorawan-stack/files/4629520/certs.zip)

I'm still using the 'acme' directory for the certificates but in the future it will be on the 'certs' directory. Just a side annotation.

The directory and the certificates have 886 as owner:

```
sudo chown -R 886:886 acme
```

The certificates have been tested and seems to be ok:

```
$ openssl verify -CAfile ca.pem cert.pem
cert.pem: OK
$ openssl x509 -in cert.pem -text -noout | grep -A 1 Identifier
            X509v3 Subject Key Identifier: 
                D1:04:74:3C:F5:40:F9:E6:E5:43:CF:39:B7:9F:35:C8:33:20:10:24
            X509v3 Authority Key Identifier: 
                keyid:63:F4:C1:63:B7:07:FB:63:80:02:70:21:C9:93:B6:29:B9:C2:7A:5E
$ openssl x509 -in cert.pem -text -noout
```

<details>
<summary>Command output (hidden to improve readability)</summary>

```
$ openssl x509 -in cert.pem -text -noout
Certificate:
    Data:
        Version: 3 (0x2)
        Serial Number: 1 (0x1)
        Signature Algorithm: sha256WithRSAEncryption
        Issuer: C = US, ST = TestState, L = TestCity, O = GeneralCA, CN = GCA
        Validity
            Not Before: May 14 06:44:01 2020 GMT
            Not After : Sep 15 06:44:01 3019 GMT
        Subject: C = US, ST = TestState, L = TestCity, O = TestLab, CN = localhost
        Subject Public Key Info:
            Public Key Algorithm: rsaEncryption
                RSA Public-Key: (2048 bit)
                Modulus:
                    00:d6:bc:47:7b:0c:61:ae:f2:5e:09:47:48:20:bf:
                    16:3b:67:43:29:f3:7a:6e:9e:76:0f:60:1b:91:ea:
                    7c:69:53:d5:9e:6d:79:bd:13:a5:e0:60:6e:fb:d7:
                    ae:fd:31:47:aa:01:85:c8:f9:9f:6c:7b:3c:c8:4a:
                    2e:2e:d4:b7:cc:de:1d:80:64:5d:e0:57:d0:74:3e:
                    4f:a5:b2:ef:49:99:c7:94:b3:e1:11:a2:0e:ab:20:
                    01:11:ac:1e:24:c9:7d:5a:5d:19:79:99:09:c7:e5:
                    fa:4c:47:87:c6:49:98:93:e0:5f:23:7c:19:c1:2c:
                    cd:2e:77:03:51:6d:aa:3a:bc:0b:aa:1f:31:ab:52:
                    1b:27:ea:bd:cc:f0:fd:2a:02:49:77:ff:cf:65:bb:
                    bf:07:d8:f2:6f:2e:a8:3f:a1:bb:2d:e6:53:fb:8a:
                    5b:c2:bc:b4:e0:43:32:c6:c3:9f:d4:6b:8f:d3:68:
                    63:f8:f9:4d:55:43:db:3f:76:d9:43:cb:9e:3f:db:
                    ad:99:85:aa:94:18:c7:99:d1:8b:ef:ff:bd:f6:8b:
                    90:64:35:f7:20:2e:11:22:21:0e:03:4e:11:c6:0b:
                    6a:2b:88:b3:e9:7f:4b:a6:f5:d6:50:d8:fe:c7:81:
                    8f:46:8b:de:c6:97:8f:24:19:fb:ae:af:11:39:58:
                    0c:df
                Exponent: 65537 (0x10001)
[... 385 lines cut ...]
```
$ sudo docker exec -it --user root tts_stack_1 /usr/sbin/update-ca-certificates
WARNING: ca-certificates.crt does not contain exactly one certificate or CRL: skipping
```

After redoing all the tests, the error stand still.

### Modifying Root CA

Also tried changing the ttn-lw-stack.yml file:

```
tls:
  source: file
  root-ca: /run/secrets/cert.pem
  #root-ca: /run/secrets/ca.pem
  certificate: /run/secrets/cert.pem
  key: /run/secrets/key.pem
  # insecure-skip-verify: true
```

And started all over again. Same "Forbidden token exchange refused" error.

---
[certs.zip](https://github.com/TheThingsNetwork/lorawan-stack/files/4629520/certs.zip)

pablorq, 2020-05-15:
### Insecure Mode ON

Modifying ttn-lw-stack.yml file to enable the *insecure mode*:

```
tls:
  source: file
  #root-ca: /run/secrets/cert.pem
  root-ca: /run/secrets/ca.pem
  certificate: /run/secrets/cert.pem
  key: /run/secrets/key.pem
  insecure-skip-verify: true
```

And started all over again: remove container, clear cookies, remove browser certificate, and start from zero.

This time the browser didn't show the certificate issue nor installed any certificate, as expected.

But after admin login, the same "Forbidden token exchange refused" error.

Can it be any clue for what's going on?

johanstokking, 2020-05-15:
From the container, can you:

```bash
$ curl -cacert /run/secrets/ca.pem https://localhost
```

Also, do you have the server logs? You can enable debug logs in the yml with:

```yaml
log:
  level: debug
```

pablorq, 2020-05-15:
Added the logging to ttn-lw-stack.yml file.

Tested in the browser against 'http://localhost' and 'https://localhost'

<details>
<summary>Log data</summary>

```
$ sudo docker-compose up
tts_cockroach_1 is up-to-date
tts_redis_1 is up-to-date
Creating tts_stack_1 ... done
Attaching to tts_cockroach_1, tts_redis_1, tts_stack_1
cockroach_1  | *
cockroach_1  | * WARNING: RUNNING IN INSECURE MODE!
cockroach_1  | * 
cockroach_1  | * - Your cluster is open for any client that can access <all your IP addresses>.
cockroach_1  | * - Any user, even root, can log in without providing a password.
cockroach_1  | * - Any user, connecting as root, can read or write any data in your cluster.
cockroach_1  | * - There is no network encryption nor authentication, and thus no confidentiality.
cockroach_1  | * 
cockroach_1  | * Check out how to secure your cluster: https://www.cockroachlabs.com/docs/v20.1/secure-a-cluster.html
cockroach_1  | *
cockroach_1  | *
cockroach_1  | * WARNING: running 'cockroach start' without --join is deprecated.
cockroach_1  | * Consider using 'cockroach start-single-node' or 'cockroach init' instead.
cockroach_1  | *
cockroach_1  | *
cockroach_1  | * WARNING: neither --listen-addr nor --advertise-addr was specified.
cockroach_1  | * The server will advertise "63a2432e5609" to other nodes, is this routable?
cockroach_1  | * 
cockroach_1  | * Consider using:
cockroach_1  | * - for local-only servers:  --listen-addr=localhost
cockroach_1  | * - for multi-node clusters: --advertise-addr=<host/IP addr>
cockroach_1  | * 
cockroach_1  | *
cockroach_1  | CockroachDB node starting at 2020-05-15 09:34:33.394488679 +0000 UTC (took 10.0s)
cockroach_1  | build:               CCL v20.1.0 @ 2020/05/05 00:07:18 (go1.13.9)
cockroach_1  | webui:               http://63a2432e5609:26256
cockroach_1  | sql:                 postgresql://root@63a2432e5609:26257?sslmode=disable
cockroach_1  | RPC client flags:    /cockroach/cockroach <client cmd> --host=63a2432e5609:26257 --insecure
cockroach_1  | logs:                /cockroach/cockroach-data/logs
cockroach_1  | temp dir:            /cockroach/cockroach-data/cockroach-temp185873162
cockroach_1  | external I/O path:   /cockroach/cockroach-data/extern
cockroach_1  | store[0]:            path=/cockroach/cockroach-data
cockroach_1  | storage engine:      rocksdb
cockroach_1  | status:              initialized new cluster
cockroach_1  | clusterID:           266affcb-7bee-4ed6-8665-ce8b1e6e3f5c
cockroach_1  | nodeID:              1
redis_1      | 1:C 15 May 2020 09:34:23.404 # oO0OoO0OoO0Oo Redis is starting oO0OoO0OoO0Oo
redis_1      | 1:C 15 May 2020 09:34:23.404 # Redis version=6.0.1, bits=64, commit=00000000, modified=0, pid=1, just started
redis_1      | 1:C 15 May 2020 09:34:23.404 # Configuration loaded
redis_1      | 1:M 15 May 2020 09:34:23.407 * Running mode=standalone, port=6379.
redis_1      | 1:M 15 May 2020 09:34:23.407 # WARNING: The TCP backlog setting of 511 cannot be enforced because /proc/sys/net/core/somaxconn is set to the lower value of 128.
redis_1      | 1:M 15 May 2020 09:34:23.407 # Server initialized
redis_1      | 1:M 15 May 2020 09:34:23.407 # WARNING overcommit_memory is set to 0! Background save may fail under low memory condition. To fix this issue add 'vm.overcommit_memory = 1' to /etc/sysctl.conf and then reboot or run the command 'sysctl vm.overcommit_memory=1' for this to take effect.
redis_1      | 1:M 15 May 2020 09:34:23.407 # WARNING you have Transparent Huge Pages (THP) support enabled in your kernel. This will create latency and memory usage issues with Redis. To fix this issue run the command 'echo never > /sys/kernel/mm/transparent_hugepage/enabled' as root, and add it to your /etc/rc.local in order to retain the setting after a reboot. Redis must be restarted after THP is disabled.
redis_1      | 1:M 15 May 2020 09:34:23.408 * Ready to accept connections
stack_1      |   INFO Setting up core component               
stack_1      |  DEBUG Loaded manifest.yaml                     mount=/assets namespace=web path=/srv/ttn-lorawan/public
stack_1      |  DEBUG Serving static assets                    mount=/assets namespace=web path=/srv/ttn-lorawan/public
stack_1      |   INFO Setting up Identity Server              
stack_1      |   INFO Setting up Gateway Server               
stack_1      |   INFO Setting up Network Server               
stack_1      |   INFO Setting up Application Server           
stack_1      |   INFO Setting up Join Server                  
stack_1      |   INFO Setting up Console                      
stack_1      |   INFO Setting up Gateway Configuration Server 
stack_1      |  DEBUG Loaded TLS certificate                  
stack_1      |   INFO Setting up Device Template Converter    
[... 132 lines cut ...]
stack_1      |   INFO Request handled                          duration=418.07µs method=GET namespace=web remote_addr=192.168.192.1:33898 request_id=01E8BVH01F3QR2DPTYVREYCCK9 response_size=25348 status=200 url=https://localhost/assets/source-sans-pro-v13-latin_latin-ext-700.639c2738552a0376c91e7d485e476fda.woff2

```
</details>

Testing curl inside the container:

```
$ sudo docker exec -it tts_stack_1 ash

/ $ curl -cacert /run/secrets/ca.pem https://localhost
curl: (3) URL using bad/illegal format or missing URL
curl: (7) Failed to connect to localhost port 443: Connection refused

/ $ cat /etc/hosts
127.0.0.1	localhost
::1	localhost ip6-localhost ip6-loopback
fe00::0	ip6-localnet
ff00::0	ip6-mcastprefix
ff02::1	ip6-allnodes
ff02::2	ip6-allrouters
192.168.176.4	c639387cb65b
```

Maybe using 'localhost' in the certificate can do a cross routing between host 'localhost' and container 'localhost'.

jpmeijers, 2020-05-15:
I'm having exactly the same issue on a new v3.8.1 install using docker-compose on my local computer, and using self-signed certificates for "localhost".

pablorq, 2020-05-15:
> I'm having exactly the same issue on a new v3.8.1 install using docker-compose on my local computer, and using self-signed certificates for "localhost".

I was trying some tests with other no-localhost certificates, but docker is messing up the latest image, so I will have to wait to pull the **real** latest image.

pablorq, 2020-05-15:
Finally I found how to get a new docker image until the 'latest' get updated:

```
docker pull thethingsnetwork/lorawan-stack:3.8
```
So now I'm using v3.8 docker file.

pablorq, 2020-05-15:
### Certificates for tts-server

To avoid any cross routing to 'localhost' from the hosts nor the container, I've configured a new server name: tts-server.

The new certificates for tts-server are:

[certs.zip](https://github.com/TheThingsNetwork/lorawan-stack/files/4635315/certs.zip)

After installed the new certificates, I added a new entry to my internal DNS server:

```
192.168.1.134	tts-server
```

On the host `ping tts-server` works as expected.

Reinstall all over again.

Testing routing to 'tts-server' inside the container:

```
$ sudo docker exec -it --user root tts_stack_1 ping tts-server
PING tts-server (192.168.1.134): 56 data bytes
64 bytes from 192.168.1.134: seq=0 ttl=64 time=0.182 ms
64 bytes from 192.168.1.134: seq=1 ttl=64 time=0.157 ms
64 bytes from 192.168.1.134: seq=2 ttl=64 time=0.159 ms
64 bytes from 192.168.1.134: seq=3 ttl=64 time=0.212 ms
^C
--- tts-server ping statistics ---
4 packets transmitted, 4 packets received, 0% packet loss
round-trip min/avg/max = 0.157/0.177/0.212 ms
```

At this point the host and the containers can route (ping) to 'tts-server'.

### Testing the tts-server certificates

Verifying curl inside the container:

```
$ sudo docker exec -it tts_stack_1 ash
/ $ curl -cacert /run/secrets/ca.pem https://tts-server
curl: (3) URL using bad/illegal format or missing URL
curl: (60) SSL certificate problem: unable to get local issuer certificate
More details here: https://curl.haxx.se/docs/sslcerts.html

curl failed to verify the legitimacy of the server and therefore could not
establish a secure connection to it. To learn more about this situation and
how to fix it, please visit the web page mentioned above.
```

On the browser, the same 'Forbidden token exchange refused' error.

<details>
<summary>Log data</summary>

```
$ sudo docker-compose up
tts_cockroach_1 is up-to-date
tts_redis_1 is up-to-date
Creating tts_stack_1 ... done
Attaching to tts_cockroach_1, tts_redis_1, tts_stack_1
cockroach_1  | *
cockroach_1  | * WARNING: RUNNING IN INSECURE MODE!
cockroach_1  | * 
cockroach_1  | * - Your cluster is open for any client that can access <all your IP addresses>.
cockroach_1  | * - Any user, even root, can log in without providing a password.
cockroach_1  | * - Any user, connecting as root, can read or write any data in your cluster.
cockroach_1  | * - There is no network encryption nor authentication, and thus no confidentiality.
cockroach_1  | * 
[... 165 lines cut ...]
stack_1      |   INFO Request handled                          duration=565.658µs method=GET namespace=web remote_addr=192.168.1.134:46568 request_id=01E8CF0VCKMCSPGVVA3P2HX34F response_size=25656 status=200 url=https://tts-server/assets/source-sans-pro-v13-latin_latin-ext-regular.80c998aa03640281d556814a14e1d9c8.woff2
stack_1      |   INFO Request handled                          duration=8.938824ms method=GET namespace=web remote_addr=192.168.1.134:46568 request_id=01E8CF0VH70J60818KJB97P8H9 response_size=0 status=302 url=https://tts-server/console/login/ttn-stack?next=/
stack_1      |  DEBUG Run database query                       duration=2ms namespace=identityserver query=SELECT users.id FROM "users" LEFT JOIN accounts ON accounts.account_type = $1 AND accounts.account_id = users.id WHERE "users"."deleted_at" IS NULL AND ((accounts.uid = $2)) ORDER BY "users"."id" ASC LIMIT 1 rows=1 source=store.go:59 values=[user admin]
stack_1      |  DEBUG Run database query                       duration=2ms namespace=identityserver query=SELECT * FROM "user_sessions"  WHERE ("user_sessions"."id" = $1) AND ("user_sessions"."user_id" = $2) rows=1 source=user_session_store.go:88 values=[964664c9-69df-4484-99a6-bac502aba47c 5db01331-ba8a-4469-ac05-cbd72dfe1e45]
stack_1      |  DEBUG Run database query                       duration=2ms namespace=identityserver query=SELECT * FROM "clients"  WHERE "clients"."deleted_at" IS NULL AND ((client_id = $1)) ORDER BY "clients"."id" ASC LIMIT 1 rows=1 source=client_store.go:113 values=[console]
stack_1      |  DEBUG Run database query                       duration=1ms namespace=identityserver query=SELECT * FROM "attributes"  WHERE ("entity_id" IN ($1) AND "entity_type" = $2) ORDER BY "attributes"."id" ASC rows=0 source=client_store.go:113 values=[63eae6ce-9d48-4ae2-984b-a2184cb44f6f client]
stack_1      |  DEBUG Run database query                       duration=2ms namespace=identityserver query=SELECT clients.id FROM "clients"  WHERE "clients"."deleted_at" IS NULL AND ((client_id = $1)) ORDER BY "clients"."id" ASC LIMIT 1 rows=1 source=store.go:59 values=[console]
stack_1      |  DEBUG Run database query                       duration=4ms namespace=identityserver query=SELECT users.id FROM "users" LEFT JOIN accounts ON accounts.account_type = $1 AND accounts.account_id = users.id WHERE "users"."deleted_at" IS NULL AND ((accounts.uid = $2)) ORDER BY "users"."id" ASC LIMIT 1 rows=1 source=store.go:59 values=[user admin]
stack_1      |  DEBUG Run database query                       duration=2ms namespace=identityserver query=SELECT * FROM "client_authorizations"  WHERE ("client_authorizations"."client_id" = $1) AND ("client_authorizations"."user_id" = $2) ORDER BY "client_authorizations"."id" ASC LIMIT 1 rows=1 source=oauth_store.go:104 values=[63eae6ce-9d48-4ae2-984b-a2184cb44f6f 5db01331-ba8a-4469-ac05-cbd72dfe1e45]
[... cut ...] ttps://tts-server/assets/ttn-console-bg.498252edc489187693a4e32162260925.png
stack_1      |   INFO Request handled                          duration=578.826µs method=GET namespace=web remote_addr=192.168.1.134:46568 request_id=01E8CF0YTR3QYS3NY1P0KZ69VZ response_size=25348 status=200 url=https://tts-server/assets/source-sans-pro-v13-latin_latin-ext-700.639c2738552a0376c91e7d485e476fda.woff2
```
</details>

---
[certs.zip](https://github.com/TheThingsNetwork/lorawan-stack/files/4635315/certs.zip)

johanstokking, 2020-05-18:
Thanks. Yes we pulled 3.8.x for now as we have some issues. So latest stable is 3.7.2.

Indeed, I'm quite sure that `localhost` is confusing here. I don't think it is sensible to set up the stack using Docker for `localhost`. It would work fine in a non-Docker environment, though.

> / $ curl -cacert /run/secrets/ca.pem https://tts-server
> curl: (3) URL using bad/illegal format or missing URL
> curl: (60) SSL certificate problem: unable to get local issuer certificate
> More details [...]

So this is the issue really. If `curl` can't trust the server cert with the CA, there's something wrong with the presented cert, trust chain or CA file. Can you check with `openssl s_client -connect tts-server:443` and see what happens?

kschiffer, 2020-06-09:
Closing this for now. We introduced some fixes to the CSRF protection recently, so I assume that this is fixed as well. If not, please reopen the issue.

closed 2020-06-09 by kschiffer