TheThingsNetwork/lorawan-stack #2524: Downlink URL in webhook header uses "localhost"

jpmeijers, 2020-05-12:
#### Summary

The headers received from a webhook integration contains the URLs one should use to send a downlink. Example:

```
array (
  'Host' => '<redacted>',
  'User-Agent' => 'ttn-lw-application-server/3.8.1',
  'Content-Length' => '1047',
  'Authorization' => '<redacted>',
  'Content-Type' => 'application/json',
  'Experiment' => 'my-experiment',
  'X-Downlink-Apikey' => 'NNSXS.<redacted>',
  'X-Downlink-Push' => 'http://localhost:1885/api/v3/as/applications/jpm-crickets/webhooks/ttnmapper/devices/cricket-001/down/push',
  'X-Downlink-Replace' => 'http://localhost:1885/api/v3/as/applications/jpm-crickets/webhooks/ttnmapper/devices/cricket-001/down/replace',
  'Accept-Encoding' => 'gzip',
)
```

The URLs given are `http://localhost:1885`, but this should be `http://server-name:1885`.

#### Steps to Reproduce

1. Enable a webhook integration
2. Send messages from a device
3. Look at the headers received on the webhook endpoint

#### What do you see now?

The downlink URL uses 'loclahost'

#### What do you want to see instead?

The downlink URL should use the correct server name.

#### Environment

The Things Stack v3.8.1

#### How do you propose to implement this?

The correct server name should be used.

#### Can you do this yourself and submit a Pull Request?

No

johanstokking, 2020-05-14:
Did you set `as.webhooks.downlinks.public{-tls}-address` ? See https://thethingsstack.io/v3.8.1/reference/configuration/application-server/

jpmeijers, 2020-05-14:
Yes that is most likely the reason. I must have missed that setting when I originally set the stack up. I see that the config options are now split into a separate file, which was different from when I set it up. I'll redo the setup and if that works I'll close this issue.

johanstokking, 2020-05-19:
@jpmeijers any news on this one? Hope it works!

jpmeijers, 2020-05-19:
I've added this to my docker-compose.yml under the stack environment:
```
      - TTN_LW_AS_WEBHOOKS_DOWNLINKS_PUBLIC_TLS_ADDRESS=https://lns.ttnmapper.org:1885/api/v3
      - TTN_LW_AS_WEBHOOKS_DOWNLINKS_PUBLIC_ADDRESS=http://lns.ttnmapper.org:1885/api/v3
```

After that `docker-compose down` and `docker-compose up -d`. On the webhook endpoint side the headers still looks like this:
```
array (
  'Host' => '<redacted>',
  'User-Agent' => 'ttn-lw-application-server/3.8.1',
  'Content-Length' => '1046',
  'Authorization' => 'test@jpmeijers.com',
  'Content-Type' => 'application/json',
  'Experiment' => 'my-experiment',
  'X-Downlink-Apikey' => 'NNSXS.<redacted>',
  'X-Downlink-Push' => 'http://localhost:1885/api/v3/as/applications/jpm-crickets/webhooks/ttnmapper/devices/cricket-001/down/push',
  'X-Downlink-Replace' => 'http://localhost:1885/api/v3/as/applications/jpm-crickets/webhooks/ttnmapper/devices/cricket-001/down/replace',
  'Accept-Encoding' => 'gzip',
)
```

johanstokking, 2020-05-19:
Ah, we have a mistake in the docs; should be `downlink` (not `downlinks`)

jpmeijers, 2020-05-20:
I changed the env vars to:
```
      - TTN_LW_AS_WEBHOOKS_DOWNLINK_PUBLIC_TLS_ADDRESS=https://lns.ttnmapper.org/api/v3
      - TTN_LW_AS_WEBHOOKS_DOWNLINK_PUBLIC_ADDRESS=http://lns.ttnmapper.org/api/v3
```

The webhook headers now looks like this, and the URLs specified seems to work to queue a downlink:
```
array (
  'Host' => '<redacted>',
  'User-Agent' => 'ttn-lw-application-server/3.8.1',
  'Content-Length' => '1053',
  'Authorization' => ':D',
  'Content-Type' => 'application/json',
  'Experiment' => 'my-experiment',
  'X-Downlink-Apikey' => 'NNSXS.<redacted>',
  'X-Downlink-Push' => 'https://lns.ttnmapper.org/api/v3/as/applications/jpm-crickets/webhooks/ttnmapper/devices/cricket-001/down/push',
  'X-Downlink-Replace' => 'https://lns.ttnmapper.org/api/v3/as/applications/jpm-crickets/webhooks/ttnmapper/devices/cricket-001/down/replace',
  'Accept-Encoding' => 'gzip',
)
```

On a general note, why is it required to have to configure all of these minor settings just to get the stack to work fully. Can't we rather have a "DEFAULT_HOST" variable that will automatically fill all these settings with a probable correct setting? In other words can't I configure "lns.ttnmapper.org" in a single place, and all the other configs will use that, unless I specifically override them?

closed 2020-05-20 by jpmeijers

johanstokking, 2020-05-20:
Great that it worked.

Please upvote and subscribe to https://github.com/TheThingsNetwork/lorawan-stack/issues/1230

PR #2571 (2020-05-25): Fix typo in AS configuration doc

#### Summary

Closes #2524 

#### Changes

- Change `downlinks` to `downlink`

#### Notes for Reviewers

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API, storage, configuration and CLI, according to the compatibility commitments in `README.md`. The target branch is set to `master` if the changes are fully compatible with existing API, database, configuration and CLI.
- [] Testing: The changes are covered with unit tests. The changes are t [... cut]