TheThingsNetwork/lorawan-stack #2529: Console unclarities

laurensslats, 2020-05-13:
#### Summary
Overview of features that need to be clarified in the Console.

...

#### Why do we need this?
Improve UI & UX
...

#### Overview

- [ ] `No-prefix` - Unclear what it is. It's also not possible to be set.
  [image]

- [ ] `Attributes` - Unclear what it is and how to set it.
  [image]

- [ ] External Join Server - unclear why this is enabled by default
  [image]

- [ ] Unclear if/how Location data of devices is used. 
  [image]

- [ ] Frequency plans with/without `(TTN)`
  [image]

johanstokking, 2020-05-14:
Thanks. @laurensslats what exactly is open for discussion?

kschiffer, 2020-05-21:
> - [ ] External Join Server - unclear why this is enabled by default
  *Description / link to documentation needed*
  [image]
Any info on this @johanstokking / @htdvisser ?

htdvisser, 2020-05-25:
I don't know. To me it also doesn't make sense to use external join servers by default.

johanstokking, 2020-05-25:
This is too much wishful thinking on my side I think. Let's change the default to not use external JS, unless the majority of new devices are indeed provisioned on a Join Server already.

kschiffer, 2020-05-25:
cc @bafonins

johanstokking, 2020-08-24:
Closing; looks like all issues are addressed individually. If not, please file the concerning issues. This umbrella issue is just a list of issues which makes it hard to plan and triage, so we close it.

closed 2020-08-24 by johanstokking