TheThingsNetwork/lorawan-stack #2534: Remove leading slash from PubSub subtopics, standardize Pub/Sub

benolayinka, 2020-05-13:
#### Summary

- Remove leading slashes from subtopic prompts (Adding slashes to topics produces strange output, and makes these integrations unusable. Slashes are automatically appended somewhere in the plumbing when they need to be and NATS doesn't use slashes anyway)
- Change PubSub and Pub-sub to Pub/Sub

#### Steps to Reproduce

Use the PubSub integration in the console

#### What do you see now?

**Topic slashes produce buggy topics**
Doing this (see prompt):
[image]

produces this (should be base-topic.nack) :
`[#27] Received on [base-topic./nack]`

**Inconsistent pub-sub / PubSub:**
[image]
[image]

#### What do you want to see instead?

- Consistent Pub/Sub label
- Remove leading slash from sub topic prompt

#### Environment

TTS V3.8

#### How do you propose to implement this?

#### Can you do this yourself and submit a Pull Request?

@kschiffer

kschiffer, 2020-05-13:
This is addressed via https://github.com/TheThingsNetwork/lorawan-stack/pull/2525.
Please verify and close issue if applicable.

johanstokking, 2020-05-14:
If we change everything to "Pub/Sub", include error messages and CLI as well

benolayinka, 2020-05-14:
> If we change everything to "Pub/Sub", include error messages and CLI as well

I was thinking that if we have to go through and fix `pub-sub` everywhere, we may as well standardize it the way everyone else (i.e. [google](https://cloud.google.com/pubsub/docs/overview) does it.. but I'm also okay with saving ourselves the pain of finding and replacing in the cli etc..

also if we change to Pub/Sub we have to standardize an alternate format (should be PubSub) for filenames where / is not a valid character

johanstokking, 2020-05-14:
Up to you guys, as long as it is consistent everywhere

benolayinka, 2020-05-14:
> Up to you guys, as long as it is consistent everywhere

Pub/Sub it is. PubSub in paths+urls. Will create an issue to track it down in console.

benolayinka, 2020-05-14:
Closed by #2525

closed 2020-05-14 by benolayinka