TheThingsNetwork/lorawan-stack #2542: Remove `No prefix` field if nothing is to be shown

laurensslats, 2020-05-14:
#### Summary
No-prefix - Unclear what it is and how it can be set.

...

#### Why do we need this?
Only if people set it by themselves, which only happens occasionally.

...

#### What is already there? What do you see now?

[image]

...

#### What is missing? What do you want to see?
Remove the `No prefix` field if there's nothing to show.
...

#### Environment
v3.7.2

...

#### How do you propose to implement this?
Only show it when there's a known prefix.
...

#### Can you do this yourself and submit a Pull Request?
@kschiffer 
...

bafonins, 2020-05-29:
I dont think we should hide the prefixes select.

1. We fetch prefixes from JS and depending on the users internet connection those can arrive after some delay. This can result in initially hiding the select and then showing it.
2. There is always at least one prefix available - the empty prefix, `0000000000000000`. During the last call we discussed that we want the users to be aware that empty `join_eui` is a viable,allowed option that developers can use for testing devices (cc @johanstokking)

That being said, I dont think we should hide the select.

kschiffer, 2020-05-29:
I disagree:

1.  If the Join EUI prefixes are a prerequisite to be able to enter the Join EUI, we need to fetch those before activating the Join EUI field as a whole. This is regardless of whether there actually are prefixes available or not and is a combined constraint of the whole Join EUI field (which then _can_ consist select and text-input)
2.  2.A select input element without options (or just one option) makes no sense and only confuses users, which is the reason this issue was created

laurensslats, 2020-05-29:
To consider:
No-one uses pre-fixes as of now nor requested this feature. This might have to do with the fact that nearly no-one knows what this is, nor how it can be set. 

> There is always at least one prefix available - the empty prefix, `0000000000000000`. During the last call we discussed that we want the users to be aware that empty join_eui is a viable, allowed option that developers can use for testing devices

If we go down this path, we need to provide 2 pre-fix options at any time (I guess the following?)
Default `26000000/7`
Experimental nodes: `00000000/7`

In addition we need documentation of:
1. What are pre-fixes 
2. When to use `default`, when to use `experimental nodes` and why
3. How to manually set pre-fixes 

_I'm personally not aware of the benefits of using an empty pre-fix over the default one._

If we don't have the above documentation in place, we only cause confusion and it asks resources from our support team because people will ask questions about it.

johanstokking, 2020-05-29:
> No-one uses pre-fixes as of now nor requested this feature. This might have to do with the fact that nearly no-one knows what this is, nor how it can be set.

It is designed for dedicated Join Servers, which are only addressable by a JoinEUI prefix. You would only be able to use those JoinEUI prefixes.

On a cluster-local Join Server, you don't need the prefix.

> If we go down this path, we need to provide 2 pre-fix options at any time (I guess the following?)
> Default `26000000/7`
> Experimental nodes: `00000000/7`

Yes, it would be something like DevAddrs; JoinEUI of all-zeros is experimental.

---

I think we have two separate use cases:

1. Dedicated Join Servers where it only makes sense to use a prefix. Here we need the dropdown
2. Clusted-local Join Servers. Here we don't need the prefix, and here an all-zeros JoinEUI is also supported

laurensslats, 2020-05-29:
Our GJS is part of option 1?

johanstokking, 2020-05-29:
> Our GJS is part of option 1?

The Things Industries Join Server is an example of option 1, indeed

kschiffer, 2020-08-24:
So as decided in the meeting today, the way this should work is that the field as a whole should be in a loading state until the list of JoinEUI prefixes has been loaded. To visualize that, we can either use the `loading` prop of the input, or wrap the whole field around the `<Overview />` component. I think it's a bit of a try-out matter to see what works better as visual feedback without being too distracting. I leave that up to you @benolayinka .

I think it's best to assume that there will be no prefix by default, so that should be the initial layout (which would get wrapped by the overlay component).

closed 2020-10-02 by bafonins