TheThingsNetwork/lorawan-stack #2545: Creating devices by passing JSON to CLI stdin does not work

rvolosatovs, 2020-05-14:
#### Summary

Creating devices by passing JSON to CLI stdin does not work

#### Steps to Reproduce

1. Pipe a valid device JSON to `ttn-lw-cli device create`

#### What do you see now?

```shell
$ cat dev.json
{
  "ids": {
    "device_id": "test-dev",
    "application_ids": {
      "application_id": "test-app"
    },
    "dev_eui": "0102030405060708",
    "join_eui": "0807060504030201"
  },
  "frequency_plan_id": "US_902_928_FSB_2",
  "lorawan_version": "1.0.2",
  "lorawan_phy_version": "1.0.2-b"
}
$ cat dev.json | ttn-lw-cli device create
 DEBUG Using access token (valid until 11:35AM)
error:cmd/ttn-lw-cli/commands:no_application_id (no application ID set)
$ cat dev.json | ttn-lw-cli device create --application-id 'test-app'
 DEBUG Using access token (valid until 11:35AM)
error:cmd/ttn-lw-cli/commands:no_end_device_id (no end device ID set)
$ cat dev.json | ttn-lw-cli device create --application-id 'test-app' --device-id 'test-dev'
 DEBUG Using access token (valid until 11:35AM)
error:cmd/ttn-lw-cli/commands:no_end_device_eui (no end device EUIs set)
$ cat dev.json | tr -d '\n' | ttn-lw-cli device create --application-id 'test-app' --device-id 'test-dev'
 DEBUG Using access token (valid until 11:35AM)
error:cmd/ttn-lw-cli/commands:no_end_device_eui (no end device EUIs set)
```

as you see I also tested if this be to some weird whitespace issues in the last invocation, but that did not help.

#### What do you want to see instead?

Successful `<json> | ttn-lw-cli end-device create`

#### Environment

3.8.1

#### How do you propose to implement this?

Actually read the stdin

#### Can you do this yourself and submit a Pull Request?

yes, but I'll assign @neoaggelos since he worked on this

rvolosatovs, 2020-05-14:
Setting priority to medium, since without this migration from v2 to v3 is not possible

neoaggelos, 2020-05-14:
Removing prio/medium since the migration guide is not affected.

It seems that the line that causes the issue is here:

https://github.com/TheThingsNetwork/lorawan-stack/blob/13d2f9d6ccd7ffac56b144507e3a5ac65d464717/cmd/ttn-lw-cli/internal/io/io.go#L100

And it was changed because of #1904. Without the `stat.Size() > 0` check, running the CLI without a TTY fails. With it, we cannot do `<something> | cli`.

We need to find a more appropriate check for this.

johanstokking, 2020-06-09:
@neoaggelos what is the status? Please set in progress if it is

neoaggelos, 2020-06-16:
I got good results by attempting to read one byte from the input stream and putting it back if successful, instead of the current check whether input is a pipe (This was also part of the original implementation in #1904). However, it does feel hacky and not too good.

@htdvisser Any pointers?

htdvisser, 2020-06-16:
We could do something like this:

```go
rd := bufio.NewReader(os.Stdin)
b, err := rd.Peek(1)
if err != nil {
	// Nothing to read
}
switch b[0] {
case '{':
	// It's at least one JSON object in a stream.
case '[':
	// It's a JSON array.
}
```

neoaggelos, 2020-06-20:
Closed with #2756

rvolosatovs, 2020-07-13:
https://github.com/TheThingsNetwork/lorawan-stack/pull/2756#issuecomment-657309318

rvolosatovs, 2020-07-14:
(continued from https://github.com/TheThingsNetwork/lorawan-stack/pull/2756#issuecomment-658143007)

> @linekermtl Based on the template.json file you attached, I think the error is that the template file is used incorrectly. As shown in the [documentation for executing the end-device templates](https://thethingsstack.io/v3.8.6/devices/end-device-templates/executing/), the correct command is:
> 
> ```shell
> $ cat template.json \
>   | ttn-lw-cli end-device template assign-euis <join-eui> <dev-eui> \
>   | ttn-lw-cli end-device template execute \
>   | ttn-lw-cli device create --application-id <application-id>
> ```
> 
> Also, how did you create this template? The template itself should not really contain the DevEUI.

There's no reason why `< $file` syntax should not work just as well.

E.g.:
```
cat test.json; cat test.go; go run test.go < test.json
───────┬──────────────────────────────────────────────────────────────────────────────────────────────────
       │ File: test.json
───────┼──────────────────────────────────────────────────────────────────────────────────────────────────
   1   │ {
   2   │     "test": true
   3   │ }
───────┴──────────────────────────────────────────────────────────────────────────────────────────────────
───────┬──────────────────────────────────────────────────────────────────────────────────────────────────
       │ File: test.go
───────┼──────────────────────────────────────────────────────────────────────────────────────────────────
   1   │ package main
   2   │ 
   3   │ import (
   4   │     "fmt"
   5   │     "io/ioutil"
   6   │     "os"
   7   │ )
   8   │ 
   9   │ func main() {
  10   │     b, err := ioutil.ReadAll(os.Stdin)
  11   │     if err != nil {
  12   │         panic(err)
  13   │     }
  14   │     fmt.Println(string(b))
  15   │ }
───────┴──────────────────────────────────────────────────────────────────────────────────────────────────
{
    "test": true
}
```

Also the DevEUI should not be a problem - users should be able to just e.g. pipe output of one CLI into another to create a device

neoaggelos, 2020-07-14:
Some further comments to clear out confusion here:

The `template.json` that was posted [here](https://github.com/TheThingsNetwork/lorawan-stack/pull/2756#issuecomment-657907782) is a Device Template file, which has a different format than a Device.

For reference:

<details><summary>Example device file</summary>

```js
{
    "ids": {
        "application_ids": {
            "application_id": "xxxxx"
        },
        "dev_eui": "xxxxx",
        "join_eui": "xxxxx"
    },
    "created_at": "0001-01-01T00:00:00Z",
    "updated_at": "0001-01-01T00:00:00Z",
    "attributes": {},
    "version_ids": {
        "brand_id": "xxxxx",
        "model_id": "xxxxx",
        "firmware_version": "xxxxx"
    },
    "network_server_address": "xxxxx",
    "lorawan_version": "1.0.2",
    "lorawan_phy_version": "1.0.2-b",
    "frequency_plan_id": "US_902_928_FSB_2",
    "supports_join": true
}
```
</details>

<details><summary>Example Device Template file</summary>

```js
{
    "end_device": {
        "ids": {
            "application_ids": {
                "application_id": "xxxxx"
            },
            "dev_eui": "xxxxx",
            "join_eui": "xxxxx"
        },
        "created_at": "0001-01-01T00:00:00Z",
        "updated_at": "0001-01-01T00:00:00Z",
        "attributes": {},
        "version_ids": {
            "brand_id": "xxxxx",
            "model_id": "xxxxx",
            "firmware_version": "xxxxx"
        },
        "network_server_address": "xxxxx",
        "lorawan_version": "1.0.2",
        "lorawan_phy_version": "1.0.2-b",
        "frequency_plan_id": "US_902_928_FSB_2",
        "supports_join": true
    },
    "field_mask": {
        "paths": [
            "version_ids.firmware_version",
            "version_ids.model_id",
            "frequency_plan_id",
            "lorawan_phy_version",
            "lorawan_version",
            "supports_class_c",
            "supports_join",
            "version_ids.brand_id"
        ]
    }
}
```
</details>

So, the issue @linekermtl is facing is not that CLI fails to read the input source altogether. Rather, the input is read correctly but cannot be parsed. The `ttn-lw-cli device create` command expects a device (not a device template). Maybe we should make the documentation or the error message a bit more clear in this case.

*NOTE*: The inclusion or not of the DevEUI on the template is irrelevant here.

In order to use a device template, one needs to execute it first with `ttn-lw-cli device template execute` and then pass it to the `ttn-lw-cli device create` command. The correct way to do this, is the command I posted on my original comment (which we also have in the docs):

```
$ cat template.json \
  | ttn-lw-cli end-device template assign-euis <join-eui> <dev-eui> \
  | ttn-lw-cli end-device template execute \
  | ttn-lw-cli device create --application-id <application-id>
```

linekermtl, 2020-07-14:
@rvolosatovs @neoaggelos Thank you both for the comments. Perhaps it would be helpful to clarify the differences between a device file and device template file in the documentation, because both files can be exported from an existing device and can easily become confusing. 

What's your recommended approach to bulk provisioning devices, using the device template file or device file?

neoaggelos, 2020-07-16:
> What's your recommended approach to bulk provisioning devices, using the device template file or device file?

The device templates are mostly meant to be used by device vendors. For bulk provisioning (where devices are identical apart from DevEUI and AppKey) I would suggest the following:

- Create one device (via the CLI or the Console) and configure as needed (Frequency plan, MAC settings, LoRaWAN protocol version, etc)
- Export the device to JSON using the CLI `ttn-lw-cli end-devices get <application-id> <device-id> | jq -c > base.json`
- Create a script that dynamically creates this JSON file for the rest of your devices (by setting the DevEUIs and AppKeys appropriately)
- Create all the devices at once with `cat my-devices.json | ttn-lw-cli end-devices create --application-id <application-id>`

For reference, `base.json` should look like this: `{"ids": {"device_id": "<dev1>", "dev_eui": "<dev-eui1>", "join_eui": "<join-eui>"}, "root_keys": {"app_key": {"key": "<app-key1>"}}, ...}`

And `my-devices.json` should look like this (note that it is JSON objects one after the other, not an array):

```
{"ids": {"device_id": "<dev1>", "dev_eui": "<dev-eui1>", "join_eui": "<join-eui>"}, "root_keys": {"app_key": {"key": "<app-key1>"}}, ...}
{"ids": {"device_id": "<dev2>", "dev_eui": "<dev-eui2>", "join_eui": "<join-eui>"}, "root_keys": {"app_key": {"key": "<app-key2>"}}, ...}
```

neoaggelos, 2020-07-17:
Closing, since issue is resolved.

PR #2756 (2020-06-20): Fix/stdin pipe read error

#### Summary

Closes #2545 

#### Changes

- Use a buffered reader and peak for input in order to decide whether stdin is available
- Use that buffered reader for further input (since the original reader has consumed the first byte of input)

#### Testing

```bash
$ cat file.json | ttn-lw-cli ... create        # works
$ ttn-lw-cli ..... create < file.json           # works
$ (create without tty as show in #1904)  # works
```

##### Regressions

It could be possible that since we may drop the first byte from `os.Stdin` with this, other CLI commands that expect user in [... cut]
closed 2020-07-17 by neoaggelos