TheThingsNetwork/lorawan-stack #2548: Change PubSub to Pub/Sub in Console strings and cli error messages

benolayinka, 2020-05-14:
#### Summary

With https://github.com/TheThingsNetwork/lorawan-stack/pull/2525 we standardized PubSub and Pub-sub as Pub/Sub, so we need to clean up console and cli messages to reflect this. 

So for messages and strings -> `Pub/Sub`
Internally, and in paths and urls, we can keep `PubSub`

#### Why do we need this ?

Present messages in a consistent and professional way.

#### What is already there? What do you see now?

Stuff like this

[image]

#### What is missing? What do you want to see?

#### How do you propose to document this?

#### Can you do this yourself and submit a Pull Request?

No. Putting this in backlog if anyone wants it.

johanstokking, 2020-06-09:
What is left here?

benolayinka, 2020-06-09:
> What is left here?

I haven't worked on this at all, but I created it as a note to chase down any lingering PubSub or Pub-sub strings in cli or console.

benolayinka, 2020-06-29:
Closing this issue as nothing depends on it

closed 2020-06-29 by benolayinka