TheThingsNetwork/lorawan-stack #2550: Impossible to create new accounts

laurensslats, 2020-05-15:
#### Summary
Password issue when creating a new account, making is impossible to access the network server.

This happens when:
- New tenants are created and admins want to get access
- New accounts are created
- Someone clicks on "Forgot password"
...

#### Steps to Reproduce
1. Create a new account
2. Click the link with the temporary password
3. Receive an error code
4. Click "Forgot password"
5: Receive the same error code

#### What do you see now?

Error code 7
[image]
```
{
  "code": 7,
  "message": "error:pkg/errors/web:unknown (Forbidden)",
  "details": [
    {
      "@type": "type.googleapis.com/ttn.lorawan.v3.ErrorDetails",
      "namespace": "pkg/errors/web",
      "message_format": "Forbidden",
      "attributes": {
        "message": "invalid csrf token"
      },
      "code": 7
    }
  ]
}
```

Error code 3
[image]
```
{
  "code": 3,
  "message": "error:pkg/oauth:no_user_id_password_match (incorrect password or user ID)",
  "details": [
    {
      "@type": "type.googleapis.com/ttn.lorawan.v3.ErrorDetails",
      "namespace": "pkg/oauth",
      "name": "no_user_id_password_match",
      "message_format": "incorrect password or user ID",
      "code": 3
    }
  ]
}
```

...

#### What do you want to see instead?
Normal process of creating passwords and accessing account.

...

#### Environment
v3.8.1+

...

#### How do you propose to implement this?

...

#### Can you do this yourself and submit a Pull Request?
@kschiffer 
...

johanstokking, 2020-05-15:
Probably related: https://github.com/TheThingsIndustries/lorawan-stack/issues/2118

We may have an issue with the middleware

kschiffer, 2020-05-18:
I'm a bit confused by this issue.

> Password issue when creating a new account, making is impossible to access the network server.

What does new account creation have to do with the network server here?

> This happens when:
> - New tenants are created and admins want to get access
> - New accounts are created
> - Someone clicks on "Forgot password"

Can you elaborate? Is this a chronological list of how to reproduce the issue or are you talking about an issue happening on each of these actions? And is this about the CSRF error you mention? 

#2560 should fix the CSRF errors.

As far as the password mismatch issue goes, I was not able to reproduce that but it indeed appears like a middleware issue. I'll wait for @adriansmares results in TheThingsIndustries/lorawan-stack#2118.

laurensslats, 2020-05-18:
@kschiffer 
I identified 3 scenarios when this issue is occurring. In any of the following scenarios people are not able to get access to their network server.

- New tenants are created and admins want to get access.
- New user accounts are created.
- Someone clicks on "Forgot password" and want to get access to their network using the received link

laurensslats, 2020-05-19:
Moved issue to: https://github.com/TheThingsIndustries/lorawan-stack/issues/2120

closed 2020-05-19 by laurensslats