TheThingsNetwork/lorawan-stack #2553: Optimal Protocol Buffers Struct serialization and deserialization to interface{}

adriansmares, 2020-05-15:
#### Summary

We need a way to optimally serialize and deserialize Protobuf `types.Struct` (or rather, `types.Value`)  to `interface{}`. 

#### Why do we need this?

- Application Server service data uses `types.Struct` for opaque service data communication
- We tend to use `types.Struct` more and more in our API usage
- We should avoid adhoc solutions (manually filling/parsing the fields, or using `jsonpb.Marshal` / `jsonpb.Unmarshal`).

#### What is already there? What do you see now?

3 solutions can be found in our code base:
- `pkg/gogoproto`: Used by payload formatters, currently seems to be incompatible with custom `struct`s for some reason. _probably the best solution to develop upon_
- Manually parsing the `types.Struct` field by field . _probably the worst solution - error prone_
- Serialize to `[]byte` the `types.Struct` and then deserialize to the actual struct. _not optimal, but robust_

#### What is missing? What do you want to see?

- Fix `pkg/gogoproto` to work on all types of structs/values.
- Remove the manual `types.Struct` parsing (one instance is the LoRaCloud DAS application package data).
- Remove the serialization/deserialization 'trick'.

#### Environment

077f48fb7d845560291460c6147d3d613a840d5b

#### How do you propose to implement this?

Removing the bad solution would be easy, but fixing `pkg/gogoproto` needs more investigation first (I've tried to remove the `package` name check, but it wasn't enough).

#### Can you do this yourself and submit a Pull Request?

Yes.

johanstokking, 2020-05-18:
What exactly is the problem?

adriansmares, 2020-05-18:
> What exactly is the problem?

The efficiency with which we serialize/deserialize our own Go objects to Protobufs `types.Struct`, since we currently have more than one solution, but none which are optimal. 

They are working, so I'll update this to gold-plating.

johanstokking, 2020-05-19:
Shouldn't we avoid marshaling our own types in `types.Struct`? Isn't the idea to use (nested) maps of scalar types, potentially optional via `BoolValue` etc?

closed 2020-07-07 by adriansmares