TheThingsNetwork/lorawan-stack #2567: Device claim authentication code form shows bad validation message

kschiffer, 2020-05-19:
#### Summary
The end device claim authentication code form will sometimes show unhandled validation messages:
[image]

#### Steps to Reproduce

1. Without setting a `Valid from` value, add a `Valid to` value
2. Remove the value using the `x` button
3. Blur the field
4. Observe uncaught validation error

#### What do you see now?
Unhandled validation error

#### What do you want to see instead?
No validation error when removing the date

#### Environment
v3.8.1

#### How do you propose to implement this?
Obviously a value casting error. The form schema should accept empty string values as well (and regard them as empty inputs).

#### Can you do this yourself and submit a Pull Request?
@mjamescompton

mjamescompton, 2020-05-29:
@kschiffer Pretty sure the `x` does not exist on the latest version of chrome, Laurens chrome is at Version 81.0.4044.138 (Official Build) (64-bit) and has the issue, I will try to downgrade my chrome so I can see the issue.

mjamescompton, 2020-05-29:
Adding to this issue, clicking save when you have a date in `Valid From` already, reverses date by one day on each save

mjamescompton, 2020-10-19:
@kschiffer unable to recreate error, could you check. I've downgraded chrome to version 80 so that the cross is still included in the calendar field, but not sure if the error can still be displayed

kschiffer, 2020-10-27:
Appears to have been resolved in the meantime.

closed 2020-10-27 by kschiffer