TheThingsNetwork/lorawan-stack #2576: How to connect  Ursalink Gateway via  CLI commands.

ursalinktech, 2020-05-21:
#### Summary
Ursalink Gateway support connect with TTI via LNS.
But we might need get to know the way to connect with TTI use different protocol. 
Same as the gateway in the link below:
https://thethingsstack.io/v3.8.1/gateways/ 

...

#### Why do we need this?
Customers required. 

...

benolayinka, 2020-05-26:
@KrishnaIyer 

> #### Summary
> Ursalink Gateway support connect with TTI via LNS.
> But we might need get to know the way to connect with TTI use different protocol.
> Same as the gateway in the link below:
> https://thethingsstack.io/v3.8.1/gateways/
> 
> ...
> 
> #### Why do we need this?
> Customers required.
> 
> ...

What is LNS?

General instructions for adding a gateway with the CLI are here https://thethingsstack.io/v3.8.1/getting-started/cli/create-gateway/

KrishnaIyer, 2020-05-26:
@ursalinktech: Please fill in the template of the issue.

KrishnaIyer, 2020-05-26:
@benolayinka and @ursalinktech: I'll cover this in a bigger issue: https://github.com/TheThingsNetwork/lorawan-stack/issues/2594

closed 2020-05-26 by KrishnaIyer