TheThingsNetwork/lorawan-stack #2581: Basic Station Integration: Spurious xtime jumps

beitler, 2020-05-22:
#### Summary
For background on `xtime` and its handling in lorawan-stack read issue #1709.
Spurious 2**32-jumps of the `xtime` value sent to the gateway in the `dnmsg` message can be observed. This permanently puts gateway and server out of sync. As a consequence, downlinks scheduled based on the wrong xtime value are rejected by the Gateway because they are too far in the future. 

#### Steps to Reproduce

1. Start a gateway process and schedule a first downlink before the first 32 bit rollover.
2. Schedule another downlink some time after the first one, but still before the first 32 bit rollover.

#### What do you see now?
The first downlink has correct xtime. For the second downlink the `xtime` value is 2**32 larger than expected. Here is a concrete example:

```
<CONCENTRATOR START>
  "updf" xtime=00e1000000721200 (0)
...
  "updf" xtime=00e10000049e2b28 (0)
  "updf" xtime=00e100000536c120 (0)
 "dnmsg" xtime=00e100000536c120 (1) < -- first downlink OK
  "updf" xtime=00e1000005cf6b00 (0)
...
  "updf" xtime=00e100002b5c4050 (0)
  "updf" xtime=00e100002bf4d668 (0)
 "dnmsg" xtime=00e100012bf4d668 (1) < -- second downlink xtime offset by 2**32
  "updf" xtime=00e100002c8d6c40 (0)
 "dnmsg" xtime=00e100012c8d6c40 (1) < -- all following downlinks offset by 2**32
  "updf" xtime=00e100002d260220 (0)     /
 "dnmsg" xtime=00e100012d260220 (1) < -´
...
```

Here is the complete station.log:
[station.log](https://github.com/TheThingsNetwork/lorawan-stack/files/4667368/station.log)

The xtime log can be extracted from it like this
```bash
head -n 300 station.log  | sed -n 's/[^{]*\({.*}\).*/\1/p' | jq -cr '[.msgtype, (select(.msgtype == "updf") | .upinfo.xtime), (select(.msgtype == "dnmsg") | .xtime ) ,.RxDelay ] | @csv' | awk -F, '{ printf "%8.8s xtime=%016lx (%1d)\n", $1, $2, $3 }'
```

I suspect the issue is rooted in the tracking of 32 bit rollovers of `xtime`. In this particular case apparently a 32 bit rollover was assumed to have happened some time between `0x00000536c120` and `0x00002bf4d668` (~11 minutes).

#### What do you want to see instead?

The class A downlink `xtime` should not be offset by 2**32 to the corresponding uplink.

#### Environment
Basic Station 2.0.4
TTN community network

johanstokking, 2020-05-29:
Thanks @beitler for reporting. Here we go again...

GS probably thinks the rollover already happened.

I suspect the issue is here:

https://github.com/TheThingsNetwork/lorawan-stack/blob/5930ecc8cc223df6918b3eb380fd61ca6fca43dd/pkg/gatewayserver/scheduling/clock.go#L102-L111

KrishnaIyer, 2020-07-28:
Hey @beitler, I tried to reproduce this but I couldn't. I'm using a TTIG with `2.0.4` firmware.

Here are my observations
##### Uplink/Downlin 1 (t=0)

020-07-28 11:09:39.904 [S2E:VERB] RX 868.3MHz DR5 SF7/BW125 snr=9.2 rssi=-45 xtime=0xEB0000200CEFEB - jreq MHdr=00 JoinEui=70b3:d57e:d001:2348 DevEui=80:0:400:35f DevNonce=7912 MIC=397349405
2020-07-28 11:09:41.313 [SYS:DEBU]   Free Heap: 18680 (min=16856) wifi=5 mh=7 cups=8 tc=4
2020-07-28 11:09:42.326 [S2E:DEBU] ::0 diid=16707 [ant#0] - next TX start ahead by 2s570ms
2020-07-28 11:09:44.877 [S2E:VERB] ::0 diid=16707 [ant#0] - starting TX in 19ms870us
INFO: tx_start_delay=1493 () - (0, bw_delay=, notch_delay=)
2020-07-28 11:09:44.903 [S2E:INFO] TX ::0 diid=16707 [ant#0] - on air: 868.3MHz 16.0dBm ant#0(0) DR5 SF7/BW125 frame=206FFD6F5DCD23B225D625B5..CFE145B7
2020-07-28 11:09:44.969 [S2E:DEBU] Tx done diid=16707
2020-07-28 11:09:45.009 [SYN:VERB] Time sync rejected: quality=2066 threshold=1066
2020-07-28 11:09:46.316 [SYS:DEBU]   Free Heap: 18680 (min=16856) wifi=5 mh=7 cups=8 tc=4
2020-07-28 11:09:47.113 [SYN:INFO] Time sync qualities: min=1047 q90=1066 max=2066 (previous q90=1066)

##### Uplink/Downlink 2 (t+2 mins)

```
2020-07-28 11:11:37.486 [SYN:INFO] Mean MCU drift vs SX130X#0: 2.0ppm
2020-07-28 11:11:41.385 [SYS:DEBU]   Free Heap: 18680 (min=16856) wifi=5 mh=7 cups=8 tc=4
2020-07-28 11:11:45.292 [S2E:VERB] RX 868.1MHz DR0 SF12/BW125 snr=8.2 rssi=-35 xtime=0xEB00002785876C - updf mhdr=40 DevAddr=270000C9 FCtrl=00 FCnt=1 FOpts=[] 018D3440E5 mic=-1669063468 (17 bytes)
2020-07-28 11:11:46.355 [S2E:DEBU] ::0 diid=16710 [ant#0] - next TX start ahead by 3s885ms
2020-07-28 11:11:46.388 [SYS:DEBU]   Free Heap: 18680 (min=16856) wifi=5 mh=7 cups=8 tc=4
2020-07-28 11:11:47.996 [SYN:INFO] Time sync qualities: min=1047 q90=1070 max=1092 (previous q90=1070)
2020-07-28 11:11:50.220 [S2E:VERB] ::0 diid=16710 [ant#0] - starting TX in 19ms825us
INFO: tx_start_delay=1493 () - (0, bw_delay=, notch_delay=)
2020-07-28 11:11:50.246 [S2E:INFO] TX ::0 diid=16710 [ant#0] - on air: 868.1MHz 16.0dBm ant#0(0) DR0 SF12/BW125 frame=60C900002780010001E5C1B9F9F0F1
```

##### Uplink/Downlink 3 (t+31 mins)

```
2020-07-28 11:40:56.021 [SYN:INFO] Mean MCU drift vs SX130X#0: 1.9ppm
2020-07-28 11:40:57.091 [S2E:VERB] RX 867.3MHz DR0 SF12/BW125 snr=9.2 rssi=-33 xtime=0xEB00008FF0192C - updf mhdr=40 DevAddr=270000C9 FCtrl=00 FCnt=2 FOpts=[] 01A5E8BA3A mic=-1002818444 (17 bytes)
2020-07-28 11:40:57.461 [SYS:DEBU]   Free Heap: 18680 (min=16856) wifi=5 mh=7 cups=8 tc=4
2020-07-28 11:40:58.354 [S2E:DEBU] ::0 diid=16735 [ant#0] - next TX start ahead by 3s700ms
2020-07-28 11:41:02.035 [S2E:VERB] ::0 diid=16735 [ant#0] - starting TX in 19ms821us
INFO: tx_start_delay=1493 () - (0, bw_delay=, notch_delay=)
2020-07-28 11:41:02.061 [S2E:INFO] TX ::0 diid=16735 [ant#0] - on air: 867.3MHz 16.0dBm ant#0(0) DR0 SF12/BW125 frame=60C900002780020001E55CFA681E97
```

##### Uplink/Downlink 4 (t+57 mins)

```
2020-07-28 12:06:27.594 [S2E:VERB] RX 867.5MHz DR0 SF12/BW125 snr=6.8 rssi=-30 xtime=0xEB0000EB29B3AC - updf mhdr=40 DevAddr=270000C9 FCtrl=00 FCnt=3 FOpts=[] 01CD63DCA9 mic=-1494950719 (17 bytes)
2020-07-28 12:06:28.361 [S2E:DEBU] ::0 diid=16754 [ant#0] - next TX start ahead by 4s194ms
2020-07-28 12:06:28.393 [SYS:DEBU]   Free Heap: 18680 (min=16808) wifi=5 mh=7 cups=8 tc=4
2020-07-28 12:06:32.536 [S2E:VERB] ::0 diid=16754 [ant#0] - starting TX in 19ms893us
INFO: tx_start_delay=1493 () - (0, bw_delay=, notch_delay=)
2020-07-28 12:06:32.562 [S2E:INFO] TX ::0 diid=16754 [ant#0] - on air: 867.5MHz 16.0dBm ant#0(0) DR0 SF12/BW125 frame=60C9000027800600012DF7809B5948
```

##### Uplink/Downlink 5 (t+72 mins, uint32 Rollover)

```
2020-07-28 12:21:38.959 [S2E:VERB] RX 868.5MHz DR0 SF12/BW125 snr=7.8 rssi=-33 xtime=0xEB0001217BEE6C - updf mhdr=40 DevAddr=270000C9 FCtrl=00 FCnt=4 FOpts=[] 01829998F2 mic=-124461864 (17 bytes)
2020-07-28 12:21:38.970 [SYS:DEBU]   Free Heap: 18680 (min=16808) wifi=5 mh=7 cups=8 tc=4
2020-07-28 12:21:39.622 [S2E:DEBU] ::0 diid=16767 [ant#0] - next TX start ahead by 4s292ms
2020-07-28 12:21:43.895 [S2E:VERB] ::0 diid=16767 [ant#0] - starting TX in 19ms867us
INFO: tx_start_delay=1493 () - (0, bw_delay=, notch_delay=)
2020-07-28 12:21:43.921 [S2E:INFO] TX ::0 diid=16767 [ant#0] - on air: 868.5MHz 16.0dBm ant#0(0) DR0 SF12/BW125 frame=60C9000027800700010CB548A623C0
```

##### Uplink/Downlink 6 (t+75 mins)

```
2020-07-28 12:24:59.213 [S2E:VERB] RX 867.1MHz DR0 SF12/BW125 snr=8.0 rssi=-34 xtime=0xEB00012D6BBDAC - updf mhdr=40 DevAddr=270000C9 FCtrl=00 FCnt=5 FOpts=[] 018A42BCA2 mic=2133864314 (17 bytes)
2020-07-28 12:24:59.844 [S2E:DEBU] ::0 diid=16771 [ant#0] - next TX start ahead by 4s335ms
2020-07-28 12:25:02.693 [SYN:INFO] MCU/SX130X drift stats: min: +0.0ppm  q50: +2.4ppm  q80: +2.4ppm  max: +3.3ppm - threshold q90: +2.9ppm
2020-07-28 12:25:02.698 [SYN:INFO] Mean MCU drift vs SX130X#0: 1.9ppm
2020-07-28 12:25:04.095 [SYS:DEBU]   Free Heap: 18680 (min=16808) wifi=5 mh=7 cups=8 tc=4
2020-07-28 12:25:04.160 [S2E:VERB] ::0 diid=16771 [ant#0] - starting TX in 19ms871us
INFO: tx_start_delay=1493 () - (0, bw_delay=, notch_delay=)
2020-07-28 12:25:04.186 [S2E:INFO] TX ::0 diid=16771 [ant#0] - on air: 867.1MHz 16.0dBm ant#0(0) DR0 SF12/BW125 frame=60C900002780080001D27AF42EC427
2020-07-28 12:25:04.804 [SYN:VERB] Time sync rejected: quality=2067 threshold=1070
2020-07-28 12:25:05.335 [S2E:DEBU] Tx done diid=16771
```

I couldn't get the Downlink `xtime` values but they closely follow the corresponding uplink's value (otherwise downlink wouldn't work).

johanstokking, 2020-07-29:
I think this is easiest to observe using a test case of `RolloverClock`. I can do that.

johanstokking, 2020-08-05:
Tested with a test case that;

- Uses the input values from the log
- Synchronizes the gateway concentrator timestamp on the first uplink, so that the initial "roll-over" is in sync
- Synchronizes the timestamp per entry, accounting for roll-overs (not happening here)
- Passing a timestamp 4 seconds after the uplink message (with RX1 of 1 second we get join delay 1 of 5 seconds), expecting that this is indeed 4 seconds after
- Passing the same timestamp, expecting that this is indeed the same concentrator timestamp

Test is passing

```diff
diff --git a/pkg/gatewayserver/scheduling/clock_test.go b/pkg/gatewayserver/scheduling/clock_test.go
index 423229228..3afd0ed62 100644
--- a/pkg/gatewayserver/scheduling/clock_test.go
+++ b/pkg/gatewayserver/scheduling/clock_test.go
@@ -121,3 +121,33 @@ func TestSyncWithGatewayConcentrator(t *testing.T) {
 	v := int64(clock.FromTimestampTime(0x499D5DD6)) / int64(time.Microsecond)
 	a.So(v, should.Equal, int64(0xAA499D5DD6))
 }
+
+func TestIssue2581(t *testing.T) {
+	a := assertions.New(t)
+
+	clock := &RolloverClock{}
+
+	ref := time.Now()
+	timestamps := []int64{
+		0x00e1000000721200,
+		0x00e10000049e2b28,
+		0x00e100000536c120,
+		0x00e1000005cf6b00,
+		0x00e100002b5c4050,
+		0x00e100002bf4d668,
+		0x00e100002c8d6c40,
+		0x00e100002d260220,
+	}
+	for i, xtime := range timestamps {
+		timestamp := uint32(xtime & 0xFFFFFFFF)
+		serverTime := ref.Add(time.Duration(timestamp) * time.Microsecond)
+
+		if i == 0 {
+			clock.SyncWithGatewayConcentrator(timestamp, serverTime, ConcentratorTime(time.Duration(xtime&0xFFFFFFFFFFFF)*time.Microsecond))
+		}
+		rx := clock.Sync(timestamp, serverTime)
+
+		a.So(clock.FromTimestampTime(timestamp)-rx, should.BeZeroValue)
+		a.So(clock.FromTimestampTime(timestamp+4000000)-rx, should.Equal, 4000000000)
+	}
+}
```

So unfortunately no reproduction via unit testing yet.

---

One thing to figure out still is if the timestamp passed to `FromTimestampTime()` goes slightly back in time, i.e. to use a reference of a past uplink message while a new uplink message has already been received and the clock has been synchronized.

---

One solution direction could be synchronizing the gateway concentrator clock on each uplink:

```diff
diff --git a/pkg/gatewayserver/io/basicstationlns/basicstationlns.go b/pkg/gatewayserver/io/basicstationlns/basicstationlns.go
index ff2fa3e0a..42888a946 100644
--- a/pkg/gatewayserver/io/basicstationlns/basicstationlns.go
+++ b/pkg/gatewayserver/io/basicstationlns/basicstationlns.go
@@ -334,20 +334,16 @@ func (s *srv) handleTraffic(c echo.Context) (err error) {
 		}
 	}()
 
-	var syncedConcentratorTime bool
 	recordXTime := func(timestamp int64, server time.Time) {
 		// The session is the 16 MSB.
 		atomic.StoreInt32(&sessionID, int32(timestamp>>48))
-		if !syncedConcentratorTime {
-			conn.SyncWithGatewayConcentrator(
-				// The concentrator timestamp is the 32 LSB.
-				uint32(timestamp&0xFFFFFFFF),
-				server,
-				// The Basic Station epoch is the 48 LSB.
-				scheduling.ConcentratorTime(time.Duration(timestamp&0xFFFFFFFFFF)*time.Microsecond),
-			)
-			syncedConcentratorTime = true
-		}
+		conn.SyncWithGatewayConcentrator(
+			// The concentrator timestamp is the 32 LSB.
+			uint32(timestamp&0xFFFFFFFF),
+			server,
+			// The Basic Station epoch is the 48 LSB.
+			scheduling.ConcentratorTime(time.Duration(timestamp&0xFFFFFFFFFF)*time.Microsecond),
+		)
 	}
 
 	for {
```

This basically eliminates rollover detection by GS.

johanstokking, 2020-08-06:
Found the cause. I can reproduce with a test case (in `pkg/gatewayserver/scheduling`):

```go
func TestIssue2581(t *testing.T) {
	a := assertions.New(t)

	clock := &RolloverClock{}

	timestamps := []int64{
		63331869818403100,
		63331869827372300,
		63331869837372200,
		63331869847372000,
		63331869857377000,
		63331869867371600,
		63331869877371600,
		63331869878424700,
		// snip
		63331870307366100,
		63331870317365900,
		63331870327365900,
		63331870337365800,
		63331870337361700, // Before the previous one
		63331870347365700,
		63331870357365600,
		63331870367365500,
	}
	var (
		prev      *int64
		sessionID int64
	)
	for i, xtimeIn := range timestamps {
		xtimeIn := xtimeIn
		diff := int64(0)
		if prev != nil {
			diff = xtimeIn - *prev
		}
		prev = &xtimeIn

		// timestamp can go back, but serverTime is always increasing
		timestamp := uint32(xtimeIn & 0xFFFFFFFF)
		serverTime := time.Now()

		if i == 0 {
			t.Log("Synchronizing gateway concentrator")
			sessionID = xtimeIn >> 48
			clock.SyncWithGatewayConcentrator(timestamp, serverTime, ConcentratorTime(time.Duration(xtimeIn&0xFFFFFFFFFFFF)*time.Microsecond))
		}
		rx := clock.Sync(timestamp, serverTime)
		tx := clock.FromTimestampTime(timestamp)

		t.Logf("xtimeIn=%016X tmst=%08X concentrator=%016X received=%v diff=%d", xtimeIn, timestamp, tx/1000, serverTime, diff)

		a.So(tx-rx, should.BeZeroValue)

		xtimeOut := sessionID<<48 | (int64(tx) / int64(time.Microsecond) & 0xFFFFFFFFFF)
		a.So(time.Duration(xtimeOut-xtimeIn)*time.Microsecond, should.BeZeroValue)
	}
}
```

Output:

```
=== RUN   TestIssue2581
    TestIssue2581: clock_test.go:166: Synchronizing gateway concentrator
    TestIssue2581: clock_test.go:173: xtimeIn=00E10000037CB91C tmst=037CB91C concentrator=00000000037CB91C received=2020-08-06 12:28:27.878791 +0200 CEST m=+0.009530010 diff=0
    TestIssue2581: clock_test.go:173: xtimeIn=00E100000405950C tmst=0405950C concentrator=000000000405950C received=2020-08-06 12:28:27.878962 +0200 CEST m=+0.009701407 diff=8969200
    TestIssue2581: clock_test.go:173: xtimeIn=00E10000049E2B28 tmst=049E2B28 concentrator=00000000049E2B28 received=2020-08-06 12:28:27.87897 +0200 CEST m=+0.009708853 diff=9999900
    TestIssue2581: clock_test.go:173: xtimeIn=00E100000536C0E0 tmst=0536C0E0 concentrator=000000000536C0E0 received=2020-08-06 12:28:27.878975 +0200 CEST m=+0.009713981 diff=9999800
    TestIssue2581: clock_test.go:173: xtimeIn=00E1000005CF6AE8 tmst=05CF6AE8 concentrator=0000000005CF6AE8 received=2020-08-06 12:28:27.878979 +0200 CEST m=+0.009718594 diff=10005000
    TestIssue2581: clock_test.go:173: xtimeIn=00E100000667EC50 tmst=0667EC50 concentrator=000000000667EC50 received=2020-08-06 12:28:27.878984 +0200 CEST m=+0.009723411 diff=9994600
    TestIssue2581: clock_test.go:173: xtimeIn=00E10000070082D0 tmst=070082D0 concentrator=00000000070082D0 received=2020-08-06 12:28:27.878999 +0200 CEST m=+0.009738467 diff=10000000
    TestIssue2581: clock_test.go:173: xtimeIn=00E100000710947C tmst=0710947C concentrator=000000000710947C received=2020-08-06 12:28:27.879005 +0200 CEST m=+0.009744058 diff=1053100
    TestIssue2581: clock_test.go:173: xtimeIn=00E1000020A1B4D4 tmst=20A1B4D4 concentrator=0000000020A1B4D4 received=2020-08-06 12:28:27.879009 +0200 CEST m=+0.009748619 diff=428941400
    TestIssue2581: clock_test.go:173: xtimeIn=00E10000213A4A8C tmst=213A4A8C concentrator=00000000213A4A8C received=2020-08-06 12:28:27.879014 +0200 CEST m=+0.009753177 diff=9999800
    TestIssue2581: clock_test.go:173: xtimeIn=00E1000021D2E10C tmst=21D2E10C concentrator=0000000021D2E10C received=2020-08-06 12:28:27.879018 +0200 CEST m=+0.009757585 diff=10000000
    TestIssue2581: clock_test.go:173: xtimeIn=00E10000226B7728 tmst=226B7728 concentrator=00000000226B7728 received=2020-08-06 12:28:27.879023 +0200 CEST m=+0.009761904 diff=9999900
    TestIssue2581: clock_test.go:173: xtimeIn=00E10000226B6724 tmst=226B6724 concentrator=00000001226B6724 received=2020-08-06 12:28:27.879042 +0200 CEST m=+0.009780997 diff=-4100
    TestIssue2581: doc.go:66: 
        /Users/johan/dev/src/lorawan-stack/pkg/gatewayserver/scheduling/clock_test.go:178
        '1h11m34.967296s' should have been the zero value
    TestIssue2581: clock_test.go:173: xtimeIn=00E1000023040D44 tmst=23040D44 concentrator=0000000123040D44 received=2020-08-06 12:28:27.879057 +0200 CEST m=+0.009796112 diff=10004000
    TestIssue2581: doc.go:66: 
        /Users/johan/dev/src/lorawan-stack/pkg/gatewayserver/scheduling/clock_test.go:178
        '1h11m34.967296s' should have been the zero value
    TestIssue2581: clock_test.go:173: xtimeIn=00E10000239CA360 tmst=239CA360 concentrator=00000001239CA360 received=2020-08-06 12:28:27.879067 +0200 CEST m=+0.009806553 diff=9999900
    TestIssue2581: doc.go:66: 
        /Users/johan/dev/src/lorawan-stack/pkg/gatewayserver/scheduling/clock_test.go:178
        '1h11m34.967296s' should have been the zero value
    TestIssue2581: clock_test.go:173: xtimeIn=00E100002435397C tmst=2435397C concentrator=000000012435397C received=2020-08-06 12:28:27.879106 +0200 CEST m=+0.009844876 diff=9999900
    TestIssue2581: doc.go:66: 
        /Users/johan/dev/src/lorawan-stack/pkg/gatewayserver/scheduling/clock_test.go:178
        '1h11m34.967296s' should have been the zero value
--- FAIL: TestIssue2581 (0.00s)
FAIL
exit status 1
FAIL    go.thethings.network/lorawan-stack/v3/pkg/gatewayserver/scheduling      0.287s
```

This is basically issue https://github.com/TheThingsNetwork/lorawan-stack/issues/1707. We should allow for small jumps back that are not considered roll overs.

closed 2020-08-06 by johanstokking

beitler, 2020-08-11:
Thanks Johan. You're correct. There is no guarantee that xtime is monotonically increasing between updf messages.
In Basic Station we allow backward jumps of 100ms:
https://github.com/lorabasics/basicstation/blob/bd17e53ab1137de6abb5ae48d6f3d52f6c268299/src/timesync.c#L59

PR #3036 (2020-08-06): Fix gateway concentrator clock roll over

#### Summary

Closes #2581
References #1707

#### Changes

- Synchronize concentrator timestamp when uplink messages arrive out-of-order
- Work-around for Basic Station gateways sending uplink frames with no `xtime` (https://github.com/lorabasics/basicstation/issues/74)

#### Testing

Unit testing, see test case for #2581

##### Regressions

Not expected

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API, storage, configuration and  [... cut]